prusa3d/Prusa-Firmware-Buddy #3095: [BUG] Filament load leaves stepper motor energized indefinitely

fatmann66, 2023-06-04:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3

**Printer firmware version** - 3.12.2-5713

**Original or Custom firmware** - Original 

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
   USB

**Describe the bug**
  If you load Filament  but then do not print right away extruder motor stays energized indefinitely.  

**How to reproduce**
 preform  filament load routine.
once loaded issue: preheat\cool down. 
come back 1 day later and find extruder motor at 50c temp 

**Expected behavior**
  expected that extruder motor would be disabled at end of load/unload routine.  
if not then when cooldown  is issued  I would expect "disable steppers" command to be issue.
NOT TESTED:  If heating is disabled by safety timer (IE did a filament load and walked away) would extruder be disabled when preheat safety timer is hit??

**G-code**
 N/A

**Crash dump file**
  N/A

**Video**
 N/A

fatmann66, 2023-06-04:
Apologies, as I see after submitting that it's under the wrong firmware folder.   close out if it can't be moved and I'll re-submit in correct location.

Prusa-Support, 2023-08-17:
Thanks for sharing your idea and no worries, the issue can be moved to the proper repository.
I see no specular issue opened by you in the other repository so I will proceed with moving it

Michele Moramarco
Prusa Research