prusa3d/Prusa-Firmware-Buddy #3097: [QUESTION] Next MINI/MINI+ firmware release?

reedy, 2023-06-06:
I know your busy making changes for the newer printers (MK4 and XL), but as they're running a fairly common codebase (at least from the same base repo) with the MINI, it seems a little odd that releases aren't being made for all devices at the same time. I'm sure at  least some of the changes and fixes being made and released are applicable for the MINI.

As per the release notes:

>Some parts of the code are shared with the Buddy board used in the Original Prusa MINI+.

It's nearly 6 months since the last MINI release (4.4.1 in Jan 2023), which in itself isn't an issue, but there doesn't really seem to be much of a (public?) timeline.

And in that time, there's been various minor and patch releases for the MK4 and the XL, and now a new major release in alpha.

The continuous developement is obviously great to see.

Should we expect this divergence to continue? Or, for example, when 5.0.0 is stable for the other devices, a release for the MINI will follow?

Is there any major reason that releases can't be built and pushed out for all the supported hardware being developed from the same codebase?

Thanks!

reedy, 2023-06-13:
This is especially relevant as I have release notifications turned on for this repo... Which means when other releases are made (such as https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.6.4 from today), I get notifications that I don't really care about :(

Lino77, 2023-06-19:
the current firmware has so many changes in it that it has no place on the mini, I've already tried to adapt it to the mini but unfortunately without success. Unfortunately the mini only has half the memory of the MK4 and until the Mk4 is up and running I don't think there will be anything new for the Mini. And the XL multitool and the MMU3 are still to come.

reedy, 2023-06-27:
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.7.0

This does what I would expect, having sections like

>Improvements valid for both XL and MK4:

and

>Improvements valid only for XL:

etc

Doing the same thing for the MINI/MINI+ should be doable...

mannkind, 2023-06-28:
It would be great, at the very least @Prusa-Support, if the code available actually compiled properly for the Mini/Mini+. 

Shouldn't there be a 🔴 or 🚨🔉  going off in Prusa's offices right now? The build for the Mini/Mini+ is objectively *broken*.
 
```
Building mini_release_boot
Memory region         Used Size  Region Size  %age Used
           FLASH:      961356 B       895 KB    104.90%
             RAM:       88388 B       128 KB     67.43%
       MEMORY_B1:          0 GB         0 GB
          CCMRAM:       64436 B        64 KB     98.32%
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

The only option right now seems to be to use v4.4.1 tag...

chrishilbert, 2023-07-03:
I asked it on a podcast/live stream a while back and they said the firmware team has basically stepped away from the Mini to get the MK4 and XL Multitool firmwares out-the-door.  I don't expect updates any time soon for the Mini.

reedy, 2023-09-18:
Ooh - https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha1

matt-chapman, 2023-09-19:
What a pleasant surprise! I was expecting maybe a small release to bring the MINI onto the same codebase for maintenance reasons, and then another long wait until anything else. But now we get _input shaping_?!

I am _unreasonably_ happy about this 😄

reedy, 2024-05-17:
And it seems they've even started doing simultaneous releases for all versions too - https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.1

closed 2024-05-17 by reedy

danopernis, 2024-05-17:
> And it seems they've even started doing simultaneous releases for all versions too

Can't be guaranteed, but we will do our best.

reedy, 2024-05-17:
> > And it seems they've even started doing simultaneous releases for all versions too
> 
> Can't be guaranteed, but we will do our best.

I mean that's fair. As long as they're continuing to be "regular" for all plaforms too, they don't necessarily need to be all simultaneous :)