prusa3d/Prusa-Firmware-Buddy #3098: [ENHANCEMENT] Get rid of disclaimer screen on boot up

OB1-OB2, 2023-06-07:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MK4
**Optional upgrades** - Input Shaper alpha firmware

**Is your suggested improvement related to an existing problem? Please describe.**
  Upon power up, the MK4 requires the button to be pressed to go past the disclaimer screen. I am requesting to remove the need to push the button so we can fully remotely control the printer and start a print job.

**Describe the expected functionality**
  I expect the printer to boot up after power up to a state that so that Prusalink can download a file and start the print remotely without requiring me to physically be present at the printer.

OB1-OB2, 2023-06-16:
I read in the blog that next alpha release is coming next week... please, please, implement this change so a print job can be started remotely after a cold start *without* having to be physically touching the printer.

Thanks!

OB1-OB2, 2023-06-22:
This is now an issue with 5.0beta2 as well... extremely annoying...

Sebastian1989101, 2023-06-27:
> This is now an issue with 5.0beta2 as well... extremely annoying...

This is most likely a safety regulation thing as Prusa is a European company. I assume it will be gone once the firmware reaches RC state.

Prusa-Support, 2023-09-16:
Thanks for getting in touch.

The disclaimer appears for safety reasons.
It was put in place absolutely intentionally because alpha and beta firmware versions may be unstable.

We expected many early adopters of "Prusa MK4 + IS FW" and we couldn't stress enough how important it is for the user to know that firmware may be unstable, especially now with the new totally different approach to printing speeds.

The user must accept the potential risks and unpredictable irregular behaviors that a non-final firmware may lead to.
The final version won't require such a disclaimer.
Please run recent a stable (final) firmware version if you need to avoid the disclaimer.

Michele Moramarco
Prusa Research

closed 2023-09-16 by Prusa-Support