prusa3d/Prusa-Firmware-Buddy #3101: [BUG] Setting a static IP address does not work.

mkatzpdp, 2023-06-13:
[MK4]
[Firmware 4.6.2]
[Original]

Wired Network with USB memory stick installed in printer

The setup screen for setting a static IP address does not work.

Set Setting -> Network -> Default to **Eth**
Set Setting-> Network -> Ethernet -> LAN to **Static**

Setting -> Network -> Ethernet -> IPV4 Address  -> press button and nothing happens except for a beep
Setting -> Network -> Ethernet -> IPV4 Netmask-> press button and nothing happens except for a beep
Setting -> Network -> Ethernet -> IPV4 Gateway-> press button and nothing happens except for a beep

I expect that pressing the button would allow for changing the IP Information

MathandCSNerd, 2023-06-27:
From what I can tell the UI currently doesn't appear to allow setting the values individually with the dial, but one way to set a static ip is to do so via the configuration file.

Put a file in the base directory of your flash drive named "prusa_printer_settings.ini" containing something like this (in fact it can contain only this if you want):

```
[eth::ipv4]
type=static
addr=192.168.0.42
mask=255.255.255.0
gw=192.168.0.1
```

More info can be found in the example ini here: https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/doc/prusa_printer_settings.ini

After plugging it in and doing a printer reset (I'm paranoid to I went into the system menu and selected the load settings from file option), it should use that static ip now.

<and for whenever my preliminary prusa forum post saying much the above is ever approved: I'm flyingkillerspacepixies so no I don't need to give credit ;P>

TojikCZ, 2023-08-14:
Hey, i struggled with this today. I kept loading settings in the wifi menu, yet it refused to look at my static IPs I've set. You have to go into the system menu, it won't work if you load in the Wi-Fi/Ethernet submenus.

I think that this should be made into a feature request for on printer IP editation capability.
The field itself is grayed out telling you it's not interactive, so I would not call this a bug.

TK0130, 2023-10-30:
I believe that one facet of this could be considered a bug (using:
```  [eth::ipv4]
type=static
addr=192.168.0.42
mask=255.255.255.0
gw=192.168.0.1 ``` doesn't actually work. At least not for some of us.)

but they grayed out menu may be a "future expansion"/"Feature request."

Either way, I will be watching this to see when static IP gets fixed.

TK0130, 2023-10-30:
Note that I know the SSID and password are correct. If I change the wifi setting from static to dynamic on the printer's screen, it connects properly and get assigned an IP address.

I copied the settings that are auto assigned to the static setup. (The settings also match what is auto assigned to my laptop other than the IP address so I know they are valid.)

```[network]
# Webservice for internal usage, listen on port 80
web=1
hostname=prusa
# Printer support two dns servers. If variables are not empty, they will be used
# even is DHCP pr AUTO type is set.
dns4=10.10.10.1;1.1.1.1

[wifi]
ssid=redacted
psk=redacted

[wifi::ipv4]
type=static
addr=10.10.10.18
mask=255.255.252.0
gw=10.10.10.1
```

mkatzpdp, 2023-10-30:
This method did NOT work for me.  I had to set a pseudo static IP on my DHCP server.

In my opinion, network issues are among the worst issues in an otherwise fantastic MK4.

TojikCZ, 2023-10-30:
@TK0130 one thing i do not see is the [network] section header in the ini file. Might be my eyesight, or you did not copy it. Just make sure it actually is in the setting file.
If that still does not work, i have no idea. I do not work on the MK4 fw team. I just had a simmilar issue a while back and managed to solve it

TK0130, 2024-02-15:
Bump 4 months later.
Also, issue duplicated on latest firmware version:

Printer firmware version - [5.1.3+12089]

TojikCZ, 2024-02-15:
Have you tried giving it the file you showed me without comments?

TK0130, 2024-02-15:
I have tried:

```
[network]
# Webservice for internal usage, listen on port 80
web=1
hostname=prusa
# Printer support two dns servers. If variables are not empty, they will be used
# even is DHCP pr AUTO type is set.
dns4=10.10.10.1;1.1.1.1

[wifi]
ssid=redacted
psk=redacted

[wifi::ipv4]
type=static
addr=10.10.10.18
mask=255.255.252.0
gw=10.10.10.1
```

And

```
[wifi]
ssid=redacted
psk=redacted

[wifi::ipv4]
type=static
addr=10.10.10.16
mask=255.255.252.0
```

Network is WPA2-PSK on channels 2,3,4,9,40,48,153, and 161 but it works fine in DNS.

vorner, 2024-03-12:
Are you _loading_ the ini file? The wifi wizard loads only the `[wifi]` section. All the other things (actually, _all_ the things) are loaded only from Settings -> System -> Load settings from file.

MightyPork, 2024-05-11:
Still broken on MINI with firmware 6.0.0 (duplicate bug #2367)

The config file was loaded but had no effect beside setting wifi credentials:

```
[wifi]
ssid=aaa
psk=aaa

[wifi::ipv4]
type=static
addr=192.168.0.7
mask=255.255.255.0
gw=192.168.0.1

[network]
dns4=192.168.0.9;8.8.8.8
```

Wifi says "link down" when using the static IP option, switching to DHCP makes it work with dynamic address.

danopernis, 2024-06-27:
Let me close this as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3549

closed 2024-06-27 by danopernis