prusa3d/Prusa-Firmware-Buddy #3102: [BUG]Sending file to printer via wired network from Prusa Slicer gives 404 error.

mkatzpdp, 2023-06-13:
[MK4]
[4.6.2 Firmware]
[2.5.2 Prusa Slicer]
[Original]

The MK4 is connected to the network via a wired connection set for DHCP.
There is a USB memory stick installed in the printer.

When ever I to download and print from the latest Prusa Slicer I get the following error message:

PrusaSlicer has encountered an error.

Error uploading to print host:

HTTP 404: 404: Not Found

(yes the 404: is duplicated).

After receiving this message the printer starts printing without any problem.

The error message is not expected.

Prusa-Support, 2023-06-14:
Hello, thank you for the report, and sorry about the message. While we sadly don't know the exact causes yet, I encountered another case like this and resolved it by reinstating the connection (setting up WiFi again, replugging cables). Would that work in your case? 

Let me know, in case of anything to further review, do not hesitate to ask in this thread at any time. 

Alessandro Pantaleo
Prusa Research

mkatzpdp, 2023-06-14:
Thank you.

I tried plugging and unplugging the network cable and that does not make the issue go away.

Since the printer prints anyway this is more of an annoyance but it is still a bug.

Prusa-Support, 2023-06-14:
@mkatzpdp, thank you for the response. In this case, would the issue still be there with the 2.6.0 RC1 PrusaSlicer? https://github.com/prusa3d/PrusaSlicer/releases. 
Also, though a long shot, would the newer MK4 firmware 4.6.4 change anything in this regard? https://help.prusa3d.com/downloads/mk4/firmware

I await your update, in case of anything in the meantime, do not hesitate to ask. 

Alessandro Pantaleo
Prusa Research

mkatzpdp, 2023-06-14:
This bug still exists in 4.6.4 of the firmware with 2.6.0 RC2 of the slicer.

Prusa-Support, 2023-06-15:
Hello @mkatzpdp, thank you for the update, sorry to see that the bug is still there. Reported this info to our devs, we'll see in detail and work on a fix as soon as possible! 

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2023-06-15:
Hello, I just noticed that this issue duplicates issue #3062, for that reason, I'll close this specific thread, but feel free to continue the thread on issue #3062 in case of anything. 

Alessandro Pantaleo
Prusa Research

closed 2023-06-15 by Prusa-Support