prusa3d/Prusa-Firmware-Buddy #3103: [BUG] Sending a print to the printer from Prusa Slicer when the printer is not on the main menu doesn't always start printing.

mkatzpdp, 2023-06-13:
[MK4]
[Firmware 4.6.2]
[Prusa Slicer 2.5.2]

Printer connected via wired ethernet cable using DHCP.

If the printers display is not on the home menu when a print is sent from Prusa Slicer with download and start printing enabled the printer does not automatically start printing.

shledge, 2023-06-27:
I noticed this happens when not on the main menu

01craft, 2023-07-15:
I repeatedly return to my computer to start another print, and remember that I can't send until I go back to the printer and return to the main menu.  Perhaps it could just give one of those "be sure the bed is clear!" warnings, but let you start a print even if it is at the "100% complete" screen from the last print.

Prusa-Support, 2023-10-06:
Thanks for reporting.

I may seem to understand this is an erratic behavior.
Being able to controllably reproduce the issues helps us resolve them more efficiently.
Given that the issue seems to be rather sporadic, it may be hard to track down, but I can't exclude an unstable connection from being part of the problem.

The connectivity features have been improved release after release over the last few months, and we continue working on it.
Please test the most recent firmware and let us know if you see progress or if you even consider the problem solved.

.
@OB1-OB2 Please consider opening a separate issue as a feature request, about the one-click printing preview upon booting up.

Michele Moramarco
Prusa Research

closed 2024-06-20 by github-actions[bot]