prusa3d/Prusa-Firmware-Buddy #3104: [ENHANCEMENT] Re-downloading a project form Prusa Slicer to the printer gives 409 file exists error.

mkatzpdp, 2023-06-13:
[MK4]
[Firmware 4.6.2]
[Prusa Slicer 2.5.2]

When sending a print to the printer from Prusa Slicer where the identical file name already exists on the printer causes a 409 error.

Prusa Slicer should inform the user that the file exists and offer the options of overwriting, start printing the original file or cancel.

ScottWell1, 2023-06-27:
Seems this was addressed in release 4.7.0; I no longer get the error (the file is silently overwritten).  Would still be nice to have a warning and ability to confirm/reject the overwrite, but at least the error is gone.

Prusa-Support, 2023-09-16:
Thanks for your suggestion.
However, this issue will be closed as it duplicates https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2447.

If you are experiencing this frequently because you frequently work on project reiterations, consider adding a _variable_  like `{timestamp}` to the _output file name format_ in Prusa Slicer so that filenames will always be unique (mind to dump old files every once in a while).

Michele Moramarco
Prusa Research

closed 2023-09-16 by Prusa-Support