prusa3d/Prusa-Firmware-Buddy #3105: Out of memory issue while idle (malloc: out of memory)

MrTrick, 2023-06-16:
I don't have the exact "steps to reproduce" here, but as best as I remember:

1. The printer was left on for about 1 week. (Prusa Mini+ with the wifi add-on board)
2. Maybe 3-4 prints had been sent to it via wifi during that time, printing for a small percentage of the total uptime.
3. When finished printing, I run the "unload filament" sequence.
4. The printer had been left alone for at least 24 hours prior to the crash.

[image]
[image]

Prusa-Support, 2023-06-16:
Hello, thank you for the report and sorry for the late response. We are aware of some possible causes: 
- It might happen when the browser is closed accidentally - in that case, the solution would be logging in again in Prusa Link and clicking on "storage"
- It might happen when the webpage is refreshing a few times "at a fast rate", like refreshing "a few times in a row".
- A long operation session might also be behind it

Is there a specific sequence of actions that you might have found causing the issue? In the sense, would it be reproducible with some specific sequence of actions? 

I'll close this particular thread being this a duplicate of some others, including #2802, in case of anything feel free to comment on that thread.

closed 2023-06-16 by Prusa-Support

MrTrick, 2023-06-21:
Thanks! I'm not aware of any actions or ways to reproduce it, and hasn't happened a second time yet. I'll keep an eye out.

globaltommy, 2023-08-18:
Just happened to me half way through the print.
I think it happened because I woke up my computer that has had the Prusa Link website opened. As soon as I woke up the PC the printer crashed.

I hope this helps