prusa3d/Prusa-Firmware-Buddy #3106: [BUG] mk4 - single extruder color change leaves unwanted blob of filament on print

jgreat, 2023-06-19:
**Printer type** - [MK4]

**Printer firmware version** - [5.0.0-ALPHA1+11007]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [None]

**USB drive or USB/Octoprint**
PrusaLink - saved to USB 

**Describe the bug**
After doing a filament color change with a single extruder, the MK4 seems to "touch off" in the last location before continuing the print.  This tends to leave a blob of filament on the surface of the print. 

I would guess that the MK4 is using the touch sensor to confirm Z height before continuing the print.

**How to reproduce**
   Any single extruder, color changing print

**Expected behavior**
  - Don't confirm Z height between color changes?
  - Touch off in a place where the next layer is going to print?
  - Wipe on bed before continuing print?
  - Do better filament extraction after color change to avoid filament leaks?

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.
[token_0.4n_0.2mm_PLA_MK4IS_3m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/11792911/token_0.4n_0.2mm_PLA_MK4IS_3m.zip)

**Video**
  Please attach a video. It usually helps to solve the problem.
https://youtube.com/shorts/svsMWv9M89Y

**Picture showing blob of filament**
[image]

bdwheele, 2023-06-21:
This looks to be related to where prusaslicer puts the M600, as noted in 
https://github.com/prusa3d/PrusaSlicer/issues/4432

zuidwijk, 2023-07-16:
> This looks to be related to where prusaslicer puts the M600, as noted in
> [prusa3d/PrusaSlicer#4432](https://github.com/prusa3d/PrusaSlicer/issues/4432)
> And possibly this patch:
> [jeremyreeder/PrusaSlicer@61b261c](https://github.com/jeremyreeder/PrusaSlicer/commit/61b261ccfd67a93f031b7d4b5112f3418dc95387)

Exactly this I still have in 5.0.0-alpha3. Right after the filament swap, when the nozzle travels to the last point, it also does makes a blob of the new filament and you have to watch out with that before it ruins the print.

capi92, 2023-08-22:
Same issue here on 4.7.2.
When adding wipe tower it's even worst, I have a huge layer shift : i.e. workaround is also buggy ! Might be a prusaslicer or not.
I need to stay in front of printer to understand what's happening !

newradius, 2023-09-07:
Happened in the latest version of IS [5.0.0-RC]

JacobChrist, 2023-10-04:
I see this too, image:
[image]

Front side (first layer look okay, it only happens at the start of the second layer:
[image]

Offending g-code:
[FlowBot 5 Color Single Case_0.4n_0.2mm_PLA_MK4IS_1h37m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12806556/FlowBot.5.Color.Single.Case_0.4n_0.2mm_PLA_MK4IS_1h37m.zip)

Prusa-Support, 2023-10-05:
Thanks for reporting.
I think this issue behavior should be improved on Prusa Slicer' side in the first place (https://github.com/prusa3d/PrusaSlicer/issues/4432).

About the suggested improvements for the filament change, oozing can't really be avoided so consider letting the nozzle ooze for several seconds before cleaning and continuing the print.
Nonetheless, for the time being, the issue can stay open as inspiration for future firmware enhancements.

Michele Moramarco
Prusa Research

JacobChrist, 2023-10-05:
@Prusa-Support This isn't oozing, it's a deliberate push that happens after later 1 but before the head starts moving. I think it moving the filament back too much after retraction but the G1 E code looks correct.

Prusa-Support, 2024-02-27:
This is supposedly addressed on the slicer' side with the release of PrusaSlicer version 2.7.2-alpha2.
The Candidate Release has already been published so the final release should be close.

Michele Moramarco
Prusa Research

closed 2024-05-14 by Prusa-Support