prusa3d/Prusa-Firmware-Buddy #3107: [FEATURE REQUEST] Missing time till color change / pause on MK4 4.6.4

Sebastian1989101, 2023-06-20:
When printing with multiple colors or with a pause, the MK4 does not display the time until this will happen. That was always a good information with the MK3S+, so it would be nice if this could be part of the info screen again.

BigBen2003, 2023-06-21:
The same problem also exists with Prusa LINK. No remaining time until the next break is displayed here either.

[image]

During a printing process, the MK4 only shows a remaining time until complete completion:
[image]

IftiBashir, 2023-07-13:
Second this - very useful feature that I hope is implemented quickly......

oakbuffalo13676, 2023-07-17:
Third - this would be awesome sauce.

pipcorona, 2023-09-04:
Here's another vote for this. My MK3s do this and I do a lot of prints where I manually swap filament or pause to insert magnets/nuts and this is really helpful to set timers to make sure I'm at the printers when they are ready.

n2yqt, 2023-09-05:
I just noticed this was missing. I'm wiritng the times on a white board now just as reminders.

Prusa-Support, 2023-10-04:
Thanks for reporting.
We will look into this.

Michele Moramarco
Prusa Research

Claghorn, 2023-11-03:
Time till pause for any reason, not just filament change (maybe I have a pause to embed something in print).

paulcobbaut, 2023-12-11:
How can this not be a feature? I was panicking because my brand new MK4 does not show the times until colour changes. PrusaLink also shows no info, so I thought I forgot this in PrusaSlicer... but I didn't. Please show the estimated time to the colour changes either on the LCD or in PrusaLink.

sdcervi, 2024-01-20:
Adding my vote. This is a crucial function, since I have a business where I sell parts that take inserted components like magnets.

BigBen2003, 2024-01-20:
English:
It is incomprehensible how one can forget such a feature, especially since it has already been implemented on the MK3.

German:
Es ist unbegreiflich, wie man so ein Feature vergessen kann, zumal es ja bisher beim MK3 bereits erfolgreich implementiert wurde.

Prusa-Support, 2024-02-22:
I seem to understand this request has been implemented and probably the commit is probably this one. https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/860bf7678c559be669703a0e306d083cc3dacd72

However, it won't be available until the next firmware public release, which will probably occur in several weeks or even a few months as there are big projects in progress. We don't have an estimated time but please stay tuned.

Michele Moramarco
Prusa Research

n2yqt, 2024-02-22:
> commit is probably this one. 860bf76

That looks like it! Thanks for the update.

Prusa-Support, 2024-05-11:
It seems that the last stable firmware release can close this issue as solved.

Michele Moramarco
Prusa Research

closed 2024-05-11 by Prusa-Support

GreaseMonkey88, 2024-10-03:
If I slice something in Prusa slicer and add a color change, I still don't see the time it in the display of my MK4S. Do I have to activate that somewhere?

ZevEisenberg, 2024-10-03:
It's on the first info screen, but I think still not on the info screen that shows up after the print has been running for a while (the one with an image of the completed model), which is annoying!

n2yqt, 2024-10-03:
Mine cycles through four different displays while printing. I can't
remember if it shows it there. I usually look on Prusa Connect. I have to
do some work this afternoon, so I'll check.

Which firmware version are you at?

GreaseMonkey88, 2024-10-03:
This is also how mine behaves, but if it stops at color change, there are only dashes (?), no time displayed. I'm on the latest firmware 6.1.3

Prusa-Support, 2024-12-17:
The current issue should be considered solved and should remain closed.
You may be talking about a recent issue preventing time-to-user-interaction codes from being issued correctly.
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4089#issuecomment-2501279274
It is partially fixed or, theoretically fixed on the firmware, pending further actions from Prusa Slicer team.

Michele Moramarco
Prusa Research