prusa3d/Prusa-Firmware-Buddy #3110: MK4 Filament Change Purge Time

addohm, 2023-06-22:
Can we please either control the purge time for ourselves, or you increase the purge time by 3-4 seconds?  Just about every material change on my printer requires me to hit "purge more".

Prusa-Support, 2023-06-22:
Thanks for reaching out to us.

The repository [prusa3d/Original-Prusa-i3](https://github.com/prusa3d/Original-Prusa-i3) is for Prusa i3 hardware and design problems.
This is a feature of the xBuddy firmware, so I transferred it to the relevant repository.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-06-22:
I happened to notice this is a duplicate of issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/518.
Please allow me to close this issue, it will remain linked to the already existing one. The `MK4` label will be included there too.

Michele Moramarco
Prusa Research

closed 2023-06-22 by Prusa-Support