prusa3d/Prusa-Firmware-Buddy #3111: [BUG] MK4 printer ignores filament runout when prinitng with Octoprint

ksheyman, 2023-06-22:
**Printer type** - MK4

**Printer firmware version** - Alpha 5.0.0-alpha2

**USB drive or USB/Octoprint**
  Octoprint

**Describe the bug**
  Printer fails to react to filament runout when printing with octoprint

**How to reproduce**
  Print with octoprint and run out of filament

**Expected behavior**
  Printer should recognize filament out and initiate filament change procedure

**G-code**
  Works with any gcode

ghoeffner, 2023-07-17:
Holy cow, just discovered this. Filament runout sensor does not work with octoprint, even on the latest firmware!

kevman28, 2023-07-17:
yup found out the hard way as well...keeps on printing

goyetus, 2023-07-17:
Please Prusa,  fix this or update.    I have all my printers  (mk3s+) with Octoprint.    Prusa connect is not enough at this moment to change.    Give us full Octoprint Support ,  or at least,   fix the minimun   ( filament change  and on-screen info)

Thanks!

ddanier, 2023-07-17:
Background for the problem:
Actually the printer does not "know" that it is printing when using the USB connection from Octoprint. This means it does see the filament run out, but will not trigger the filament change based on this information. The issue is that the firmware stays in main menu mode, where the commands received by USB are just executed without any "thought".

This thing is especially bad, as the printer continues to print until it cannot push the filament any further. This also means you cannot just unload the filament afterwards. I had to remove the heater block and the remove the filament from it at least twice. This is really annoying, as it takes about 15 minutes if you count in cooldown times! Currently my only way to print with spools that are nearly empty is my Mini.

Speaking of the Mini: When looking at the Mini the firmware switches to a Octoprint "in print" screen which also allows you to change certain settings (like print speed, live z etc). Meaning the MK4/XL firmware "just" has to also include such a print screen, when using the USB connection (or maybe show the screen after some GCode?). As the Mini firmware is the basis for MK4 and the XL it SHOULD be possible to port this over somehow.

Also see https://forum.prusa3d.com/forum/user-mods-enclosures-nozzles/prusa-mk4-octoprint/ for some details and observations by other users on this. This issue is bugging A LOT of people. This is the post with the most replies and views BY FAR in this category.

kevman28, 2023-07-17:
> Background for the problem: Actually the printer does not "know" that it is printing when using the USB connection from Octoprint. This means it does see the filament run out, but will not trigger [...]

great detailed information. I wanted to add, I ended up being able to load new filament through to avoid the rebuild issue you mention.

ddanier, 2023-07-24:
Just a quick note: I just watched the filament run out and then sent a M600 GCode command manually through OctoPrint. This seems to have worked, but I was staring at the printer for at least 15 minutes. Hope this still helps someone, the important bit is that you can still change the filament manually.

derlucag, 2023-07-24:
> Just a quick note: I just watched the filament run out and then sent a M600 GCode command manually through OctoPrint. This seems to have worked, but I was staring at the printer for at least 15 [...]

At this point i‘d use the printers manual controls to invoke a filament change. But thanks for the input :)

goyetus, 2023-07-25:
If mk4 is not sending the "code"  when filament sensor  runs out ,   Im not going to put  any mk4  into my  mk3s+ farm.   

It´s simply.    If they no add it,  is cause they dont want to add it.   No more excuses.  

We are free to purchase another printers.   This post is as old as 2 years  for the "mini".    

Prusa says it´s "open"  but things like this,  are worst that being a closed open source company.

fergymcferguson, 2023-08-01:
+1 Bump

chadmoller, 2023-08-24:
This is the only thing keeping me from purchasing an MK4. I've already hacked together an aftermarket sensor on another printer to make the run out work and I'm not going to do it at this price point.

This was going to be my first Prusa printer. Nevermind

Syl20-94, 2023-09-13:
problem is still activ with FW 5.0.0 RC, does someone have an idea if this is handle  ?? @Prusa-Support ??

welnanick, 2023-09-28:
They are working on it: https://twitter.com/Prusa3D/status/1705151177329959185?s=20

jakkaj, 2023-09-30:
Sweet. I am looking forward to this release!

hcooper, 2023-10-04:
For the record, the tweet says:

> Octoprint is not working the way we want on the MINI / MK4 / XL (e.g. no progress on the LCD screen, not being able to swap filament...). The problem is on our side (not in Octoprint). Our firmware team is working on resolving these issues and improving @OctoPrint3D integration.

Reacy1998, 2023-11-26:
Yesterday I installed the new FW 5.1.0 and noticed that the screen was different when printing from octoprint.
So I tested the filament runout and it seems to be working...
Has anyone else noticed this too?
[image]

hockeygoalie35, 2023-11-27:
@Reacy1998 Yes, the latest firmware has Alpha support for Octoprint. That screen is similar to what the mini has when Octoprint is connected. All in all, good news and a great step forward to Octo support.

ccle610, 2023-12-03:
Indeed, Filament Runout was noticed by Octopint in FW Rel 5.1.0 alpha and beta . . . 
Today, the official Release of 5.1.0 ist installed, Filament ran out and it was NOT recognized by Pctoprint ;-(
So, I've tested it twice, same result!
It seems, there was a Change (Bug) in the official Release?

I put in these G-Code M604, M601 and M602 for Pausing like described in Octoprint G-Code Scripts. 
I did not check it out without this Change.

closed 2024-06-20 by github-actions[bot]