prusa3d/Prusa-Firmware-Buddy #3112: [BUG] MK4 Alpha2 Firmware fails to home X/Y axis

reiuwu, 2023-06-24:
**Printer type** - MK4

**Printer firmware version** -5.0 Alpha2

**Original or Custom firmware** - Original

**Optional upgrades** - Factory Prebuilt

**USB drive or USB/Octoprint**
USB

**Describe the bug**
Starting a print using the IS profile on stable 2.6.0 slicer will start the homing process. Bed and Extruder will move to the Left/Back of printer, and continually tap against the limit. Eventually the process times out and may produce a printer collision error(error screen with no error code), or a Homing Error (prusa.io/13305 C01LK59D). 
Reset the printer config, ran the setup wizard. It failed to properly home. 

Downgraded to 5.0 Alpha 1, ran setup wizard. Did not encounter any errors, print is running without error. 

**How to reproduce**
  Flash 5.0 Alpha 2 firmware. Home Printer. 

**Expected behavior**
  Printer should home without error. 

**G-code**
N/A

mp-se, 2023-07-04:
I have the same issue on 4.7.0, here is a crash dump from that. Ive spent hours with support trying to fix my mk4 kit. 

[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/11945119/dump.zip)

mp-se, 2023-07-04:
I've now tested with the 5.0.0-alpha2 firmware and the result is also a crash. It tries a little longer before the crash this time. here is the dump from the selftest. 

[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/11945401/dump.zip)

Prusa-Support, 2023-09-26:
Thanks for reaching us.

I suspect this issue was fixed on [FW 5.0.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.0.0) (final).
Please give it a try.

However, the description of the issue matches well with mechanical problems too: obstacles on the axis, caught/hitting cables, dry rods/bearings, incorrectly tightened bearings, incorrectly tightened belts, incorrectly installed idler bearing, misaligned motor pulley... only to name a few reasons.

Please let us know if the most recent firmware worked for you but I strongly encourage you to reach out to [Customer Support](https://help.prusa3d.com/article/customer-support_2287) if this problem persists with the stable firmware release before reporting new problems.

Michele Moramarco
Prusa Research

shledge, 2023-09-26:
5.0 final seems to have fixed it for me :)

Prusa-Support, 2023-10-13:
Thanks for your feedback.
This issue will be closed now.

Michele Moramarco
Prusa Research

closed 2023-10-13 by Prusa-Support