prusa3d/Prusa-Firmware-Buddy #3113: Problém s přítlakem filamentu u MK4

DraKus6062, 2023-06-25:
Testuji novou MK4.
Při testu tisku PC blend (vysoká teplota trysky) přestal fungovat posun filamentu.
Zjištěn velmi malý přítlak filamentu na posuvné kolo.
Přitažením šroubků s pružinami se přítlak nezvýšil.
Vypadá to, že materiál (asi stále PETG), ze kterého jsou přítlačné díly tištěny, takovou teplotu nevydrží
a deformují se.
Vložením fousku fólie o tl. 0,2 mm mezi přítlačnou kladku a přítlačnou páčku jsem problém na chvíli odstranil.
Asi by to chtělo tisknout tyto díly z teplotně odolnějšího materiálu.
drahomír.kusák@dk-elvis.eu

davejenej, 2023-06-29:
Dobrý den, kontaktujte podporu Průši na jejich webových stránkách. Zde se řeší bugy softwaru.

Prusa-Support, 2023-09-19:
Thanks for reaching us.

I seem to understand this concerns the hardware so this is not the correct repository.
The correct repository is https://github.com/prusa3d/Original-Prusa-i3 however I'm afraid this is only a matter of [Extruder Idler assembly](https://help.prusa3d.com/guide/5-nextruder-assembly_434014#435942) so first thing first please contact our [Customer Support](https://help.prusa3d.com/article/customer-support_2287).

Our community is very large and spread all over the world, and our collaborators are not only Czech.
Possibly use the English language as this helps more users and collaborators to find the topic, help, or add feedback.

Michele Moramarco
Prusa Research

closed 2023-09-19 by Prusa-Support