prusa3d/Prusa-Firmware-Buddy #3114: [ENHANCEMENT] Upload Firmware to Prusalink or PrusaConnect

yasars, 2023-06-27:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MK4

**Is your suggested improvement related to an existing problem? Please describe.**

Hi Can you add an Option to install the firmware from link or connect or better online firmware update check 🗡️ 
i love your support of MK4 !! weekly updates fantastic !!

chrishilbert, 2023-06-27:
Agreed.  Allow G-CODE and BBF upload at minimum.  

Would be nice to show current version with a button next to it showing an update available on Prusa Connect.  Put a download and update button next to it in the GUI.

Prusa-Support, 2023-09-14:
Thanks for reaching out to us.

I believe the OTA firmware upgrade via Prusa Connect was enabled about 3 months ago.

[image]

At this point, the issue can be closed but please give it a go and report back.

Michele Moramarco
Prusa Researchh

closed 2023-09-14 by Prusa-Support

cmsj, 2023-09-18:
@Prusa-Support I think you should re-open this - being able to upload .bbf files through PrusaLink is IMHO a valid feature request. I do not use PrusaConnect, but I would very much like to be able to upload firmware files remotely and power cycle the printers.

Prusa-Support, 2023-09-28:
@yasars Have you checked if "'announce prerelease firmware' was enabled in the printer's Prusa Slicer settings?
[image]

@cmsj The last request instead is probably relevant for [prusa3d/Prusa-Link-Web](https://github.com/prusa3d/Prusa-Link-Web) (UI) or [prusa3d/Prusa-Link](https://github.com/prusa3d/Prusa-Link) (core) repositories, however, Prusa Link should remain the most basic interface. It is a sort of bridge between the printer and Prusa Connect and it is hosted on the printer' side so it will probably not be packed with many features.

Michele Moramarco
Prusa Research

yasars, 2023-09-28:
@Prusa-Support yes it is/was activated but no notification was received... I also think that we upload the firmware to the printer...instead of just gcose please also insert bbf...

Prusa-Support, 2023-11-17:
The possibility to _install alpha/beta versions_ of the firmware is actually not implemented and its implementation is not in the current plans.
The _announcement of prerelease firmware_ is not fully implemented either but Prusa Connect is a beta release. The option is there but it is a work in progress though.

Please consider opening a new more specific issue about the possibility of installing a prerelease firmware from remote, but opening a new issue may not be necessary because this is in the works already.

Michele Moramarco
Prusa Research

revolter, 2023-11-27:
> I believe the OTA firmware upgrade via Prusa Connect was enabled about 3 months ago.

It doesn't show up for me. Was is disabled in the meantime?

revolter, 2023-11-27:
How do I upload it?

yasars, 2023-11-27:
`go to.prusa connect and the upload button for the gcodes`

revolter, 2023-11-27:
It doesn't let me select the `prusa3d_fw_MK3S_MK3S+_3_13_2_7080.hex` file this way 🤔

yasars, 2023-11-27:
on mk4 buddy firmware is working

T0TProduction, 2023-12-02:
my mk4 and prusa connect even show me this nice interface when I try manually uploading the .bbf file
[image]
but then it ends up throwing this error:
[image]

this looks like it shouldn't be happening

T0TProduction, 2023-12-02:
In addition to that the UI only links to 5.0.1
[image]

although 5.1.0 seems to already be out? https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0

revolter, 2023-12-02:
@Prusa-Support, would you be able to help?

Prusa-Support, 2023-12-27:
The firmware can be upgraded via Prusa Connect.
Prusa Link is a basic local interface with limited functionalities and will most likely remain like that.

As far as I know, most of the firmware OTA upgrading work is made possible thanks to the advanced bootloader on Buddy 32-bit boards, on the printer' side. I can't tell if old printer models based on 8-bit boards will ever be compatible with this feature but please notice this repository is intended for Buddy 32-bit boards (MINI, MK4, MK3.9 & XL).

Michele Moramarco
Prusa Research

maybeanerd, 2023-12-27:
That comment didn't help, but since two weeks or so it just works in prusa connect again
Guess it was temporarily broken

sejef, 2024-03-06:
Hi everyone, regarding printer firmware updates, I was wondering if it was possible to add the avrdude-slic3r binary used by PrusaSlicer to the prusalink img to update the firmware locally. That would be very useful! I tried using the original avrdude but got a lot of write/read errors. Thanks!

Isegrimm666, 2024-06-28:
Hi ...

I am a newcomer to the scene and tend to use the local version of 'Prusa Link', i.e. exclusively in the local network.

I have also been looking for a way to download the latest firmware to the USB stick via Prusa Link and then reboot the printer remotely. Unfortunately, this function does not exist and it would certainly be a great advantage if it were created, as it somehow does not really correspond to the character of remote control software if I have to disconnect the printer and a USB stick for maintenance work.

Stefan