prusa3d/Prusa-Firmware-Buddy #3115: [BUG] Tool Light Option Missing

chrishilbert, 2023-06-27:
**Printer type** - XL

**Printer firmware version** - 4.7.0

**Original or Custom firmware** - Original

**Optional upgrades** - Single Tool

**USB drive or USB/Octoprint**
USB

**Describe the bug**
Settings -> User Interface is missing the Tool Light option mentioned in the release notes after upgrade.

**How to reproduce**
 Updated single tool to 4.7.0.  Ran full calibration and restarted printer.

**Expected behavior**
Show Tool Light option as mentioned.

Prusa-Support, 2023-09-09:
Thanks for repeating.
This seems to be fixed by now.
Please upgrade the firmware and try again, even please try a factory reset if the option is not listed.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-09-29:
It seems safe to close this issue now.

Michele Moramarco
Prusa Research

closed 2023-09-29 by Prusa-Support