prusa3d/Prusa-Firmware-Buddy #3117: [BUG]  Prusalink Create Folder with Chrome

mhornsby, 2023-06-28:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4]

**Printer firmware version** - [4.7.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB\

**Describe the bug**
  A clear and concise description of what the bug is.
Using Chrome the create folder option
[image]

After you cliock confirm no folder is created , the button stays on the screen and you have to refresh the browser
[image]

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

Jinra, 2023-07-21:
+1 happening on Firefox as well. This also prevents further attempts to create a folder, and also prevents you from uploading to the USB drive via prusaslicer.

Prusa-Support, 2023-11-02:
Thanks for reporting.

I don't seem to be able to reproduce the issue.
As of now, I'm able to easily create or remove folders via Prusa Link or Prusa Connect, several times, without incurring similar problems.
Not sure what went wrong in the first place and I'm not able to reproduce the problem now.

There have been various firmware and system upgrades recently and we expect more.
Please try again with the most recent firmware and consider closing the issue if you are no longer able to reproduce the problem.

Michele Moramarco
Prusa Research

mhornsby, 2023-11-02:
Hi 

I just tried on 5.0 and have the same issue. I've attached a screen recording.

The sequence is
- using the web prusalink url
- click the create folder icon
- you get a pop-up asking for the folder name
-  enter a folder name
- you cannot get out of the screen

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/18347118/4262da53-2456-4156-a98f-7aff4a37395f

Thanks
Michael

theotherjayntaylor, 2023-11-13:
I'm having the same issue.
Mk4
Firmware 5.0.1, 
Original firmware,
Using USB drive

In my Chrome debugger, it makes a call to `api/v1/files/usb/<name of folder I'm trying to create>` which never finishes.  It just stays in `Pending` state until it ends in a "too many retries" error

[image]
[image]

PhLacoude, 2023-11-22:
Exact same problem with firmware 5.0.0

Printer type - [MK4]

Printer firmware version - [5.0.0]

Original or Custom firmware - [Original]

Optional upgrades - [None]

USB drive or USB/Octoprint - [N/A: I am using the web interface http://mk4.local/#files where mk4.local is my printer on my local network]

Describe the bug - [Using Chrome and the create folder option. VPN is turned off. I have excellent connectivity to the printer from my desktop.]

theotherjayntaylor, 2023-11-28:
> I'm having the same issue. Mk4 Firmware 5.0.1, Original firmware, Using USB drive
> 
> In my Chrome debugger, it makes a call to `api/v1/files/usb/<name of folder I'm trying to create>` which never finishes. It just stays in `Pending` state until it ends in a "too many retries" error
> 

As of 5.10 firmware, this is fixed for me.

mhornsby, 2023-11-29:
Its noew creating the folder ok in 5.1.0 however its not refreshing the browser so you have to do that manually before you can see the folder. So not quite fixed.

Prusa-Support, 2023-11-30:
I'm glad to hear that FW 5.1.0 improved the situation, but the problem was not always reproducible as far as I know.
Please test a few more times to make sure the problem doesn't sporadically occur again, and let me know if the issue can be closed.

@mhornsby Your last suggestion is a quite specific feature request.
It may deserve to be posted as a separate issue so we can track problems and suggestions more efficiently.

Michele Moramarco
Prusa Research

closed 2023-12-05 by mhornsby