prusa3d/Prusa-Firmware-Buddy #3119: MK4 Insert New Filament

addohm, 2023-06-30:
When I insert filament, can we just start the process of pulling the filament through without me having to click "continue" - like how it worked on the MK3?  Its an extra step that I can't see any reason for.

Prusa-Support, 2023-06-30:
When the user inserts filament, the autoload feature will be triggered, thus a list of materials will be displayed. The user must pick a material and the printer will load and purge the filament autonomously.
On the next interaction with the user, the printer will ask if loading was successful (will also give the option to purge more).
The only scenario that I may think for the printer to ask the user to press '_continue_' is when the filament goes undetected mid-way through the loading process.
This is the behavior observed on the alpha firmware I'm testing now (FW 5.0.0).

Not sure if the issue is valid. Maybe we are simply missing clear steps for reproducibility. A video may help too.

Michele Moramarco
Prusa Research

pigskins65, 2023-06-30:
> When the user inserts filament, the autoload feature will be triggered, thus a list of materials will be displayed. The user must pick a material and the printer will load and purge the filament [...]

Not OP, but when I swap filament mid-print, the MK4 will have me click Continue after removing filament, then Continue again after I insert the new filament. Maybe that is what is being referenced?