prusa3d/Prusa-Firmware-Buddy #3121: [BUG] MK4 Random Layer Compression/Over Extrusion 5.0.0-alpha2

Mulamike, 2023-07-01:
**Printer type** - MK4 Factory Assembled - 19 Days print time

**Printer firmware version** - 5.0.0-alpha2

**Original or Custom firmware** - Original

**Optional upgrades** - Stock Filament Runout Sensor

**USB drive or USB/Octoprint** - Samsung 3.1 Bar Plus 128GB USB flash drive formatted FAT32

**Describe the bug** - Random layer compression/over extrusion using Jet Black Prusament on 3 of the same prints(Landy Mini Frame) using alpha firmware and the most recent Prusa Slicer 0.15mm Quality profile, non input shaper speed. I reverted back to 4.7.0 to test same GCODE and returned back to 5.0.0A2 to test again, both times re-doing initial printer calibration test. Stable 4.7.0 firmware does not produce these results with the same GCODE. I noticed this issue with other parts on my Landy Mini, the issue is not consistent on each print, but in different areas and does not appear sometimes.

**How to reproduce** - I'm not sure currently

**Expected behavior** - Print at least as good as 4.7.0 using non input shaper speed profiles

**G-code** - In zip

**Crash dump file** - In zip

**Video** - I can record another print if needed.

[MK4 Layer Compression.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/11926573/MK4.Layer.Compression.zip)

[MK4 LC2.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/11926575/MK4.LC2.zip)

chrishill-github, 2023-07-02:
I can confirm this behaviour. Happens with both PETG and PLA, when sliced with the normal (non-IS) profile.  Same file sliced for the MK3S+ prints fine on the MK3S+.

After some trial and error, the problem seems to start after about 8.5m of filament have been printed.  As well as the sudden onset of over-extrusion at this point, there seems to be a small change (reduction) in the extrusion width somewhat earlier, after maybe 3m (guessing) of filament.

Very noticeable on a file sliced with 8 perimeters (grey print), but still still present on one with 2 perimeters (blue print).
[image]
[image]

MK4 kit, PS 2.6.0, FW 5.0.0 Alpha2

chrishill-github, 2023-07-02:
No sign of this issue when using the IS profile:
[image]

Mulamike, 2023-07-04:
I wonder if 5.0.0-alpha3 fixes this.

chrishill-github, 2023-07-04:
Seems to.  Nothing in the release notes that addresses this issue specifically, but some changes to Pressure Advance that "also fixes all remaining issues of PA during filament manipulations".

I'm currently printing the Gcode that generated the first grey PLA print above, and it's gone past the problem layers with no sign of the issue.

It would be good to hear of your experience with Alpha3.

chrishill-github, 2023-07-05:
Looks fixed to me.
[image]

Mulamike, 2023-07-24:
Alpha3 has resolved this issue.

closed 2023-07-24 by Mulamike