prusa3d/Prusa-Firmware-Buddy #3122: [BUG] <Enter comprehensive title>

MarkusP85, 2023-07-01:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 5.0.0 Alpha 2

**Original or Custom firmware** - Original

**Optional upgrades** - none

**USB drive or USB/Octoprint**
  USB drive or Prusa Connect

**Describe the bug**
 Can't switch back to FW 4.7.0. When trying to force a FW update like discribed on page 31 of the handbook the FW update won't start and the information for the Alpha version will pop up.
**How to reproduce**
  Install the Alpha 2 and try to switch back to an older version

**Expected behavior**
 FW update will start

**G-code**
 not applcable
**Crash dump file**
not applicable

**Video**
  

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/90247386/64f8249c-cd12-4402-9e9d-e6c538260e64

Mulamike, 2023-07-01:
Try clicking the knob while it boots up, that works for me instead of holding it down.

MarkusP85, 2023-07-02:
> Try clicking the knob while it boots up, that works for me instead of holding it down.

Thank you that worked. So there is a mistake in the handbook saying ypu should hold the button down

closed 2023-07-02 by MarkusP85