prusa3d/Prusa-Firmware-Buddy #3123: [FEATURE REQUEST] Total Print Time When Complete

tbedore, 2023-07-01:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MK4]

**Describe the expected functionality**
 Same way that the MK3s+ would tell you at the end of the print, total print time 3 hrs 26 minutes. Would like to be able to see at the end of the print how long it actually took to print the file

gudnimg, 2023-07-14:
I agree with this. Maybe it can be added to this screen? There is some space underneath "Print finished" text.

[image]

murk-sy, 2023-08-02:
Duplicate of #3044

tbedore, 2023-08-02:
as this is a duplicate of #3044 I'll close the ticket :)

closed 2024-06-04 by danopernis