prusa3d/Prusa-Firmware-Buddy #3124: 5.0.0-alpha2 firmware crashes at same place during printing

Kailin12, 2023-07-03:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4]

**Printer firmware version** - [5.0.0-alpha2]

**Original or Custom firmware** - [Original alpha release]

**Optional upgrades** - [Stock]

**USB drive or USB/Octoprint**
  Print media is the stock ADATA USB stick included with printer.

**Describe the bug**
  During printing the printer's firmware will abruptly crash. It happened on different gcode files and always at the same place (different for each gcode)

**How to reproduce**
  I printed the file where the issue occured without filament and recorded it.

**Expected behavior**
I expected the print to finish normally, but the firmware crashed.

**Files**
I have included a folder with a video demonstrating the problem, the gcode file and the dump.bin file I made after the printer restarted.
[MK4 ALPHA 2 BUG.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/11940587/MK4.ALPHA.2.BUG.zip)

GuidoHendriks, 2023-07-04:
You should retry it with the latest alpha and report the result. 😇

Prusa-Support, 2023-09-27:
Thanks for reporting.

The version of the firmware 5.0.0-alpha2 solved the known cases of crashes back then however firmware crashes (BSOD or boot loops) persisted in a few cases probably in most of the alpha firmware subsequent releases.

Several firmware pre-released occurred and the firmware was improved in many ways.
Please give it a go with the final 5.0.0 release.
Consider performing a hard reset if the problem persists (it requires the firmware .bbf file on the USB flash drive).

Michele Moramarco
Prusa Researc

closed 2023-09-27 by Prusa-Support

Kailin12, 2023-09-27:
Yes this problem was persistent on older ore release versions and has since
been solved, thank you.