prusa3d/Prusa-Firmware-Buddy #3125: [FEATURE REQUEST] PrusaLink Last Print Time/Date

mhornsby, 2023-07-03:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MK4]
**Optional upgrades** - [e.g. Filament Runout Sensor]

**Is your feature request related to an existing problem? Please describe.**
  A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the expected functionality**
  A clear and concise description of expected functionality.

There's a last modified time.
[image]

Can we get a last print time. Use case I just started a print and aborted it because it didn't stick to the build plate and I need to fix that. I have about 20 files in a folder and now trying to work out which file that was I aborted.

Prusa-Support, 2023-09-23:
Thanks for your suggestion.

I'm afraid we are looking at the metadata on the file that is created on the computer's side, however, a print chronology may sound like a good idea to me. This may take up precious memory resources so should probably be limited to a few entries only.
Our developers will make their considerations.

Regardless of this, a print history and advanced statistics are already available on Prusa Connect when you print through it.

Michele Moramarco
Prusa Research

thanhtacles, 2023-12-07:
Why isn't there even the ability to sort by date of the file? The information is there, and even manually inserting the USB drive it'll show you files by date. No reason why prusalink should be inconsistent here.

closed 2024-06-20 by github-actions[bot]