prusa3d/Prusa-Firmware-Buddy #3126: [ENHANCEMENT] Printing Time improvements

inieto, 2023-07-04:
**Printer type** - [MINI+]

- "Probing point 1...16/16" could be done while heating the nozzle and bed in parallel. It's sequential actually and it take so many time

- "Probing point 1...16/16" waits the nozzle to cool down to 170ºC. It could actually be done without setting the temperature to 170ºC. Let's imagine the following example:
· First we wait from ambient temperature to 260ºC to load some filament.
· Then we have to wait for cooling down to 170ºC to make the probing point procedure
· After that, we have to wait again to heat up to 260ºC to start printing
We end up waiting 3 heat-cool-heat cycles!

- Loading filament should not wait for bed heating: we only need to wait for the nozzle to heat, not the bed.. This could save us some minutes.

shledge, 2023-07-05:
The problem is that the nozzle and bed expands under heat, so your method will give more inconsistent results. That's why nozzle is at a constant 170C and the bed is heated when levelling.

I reckon retracting the filament slightly before probing would work better in terms of saving time, because it means less time needed to do a nozzle cleaning procedure prior to levelling. The nozzle cleaning could be wiping against the bed rather than going down slowly between each point.

neophyl, 2023-07-05:
Besides this is all set and controlled by the starting gcode in your Printer Profile in the slicer.  If you don't like the order it does things or want to omit or change temps etc then go and update your profile and get it working the way you want it to.

Shledge is quite correct though on the reasons for it, repeatability of bed levelling, as the sensor is temperature dependant and even though there is some compensation it is limited in effectiveness.

closed 2023-07-06 by inieto