prusa3d/Prusa-Firmware-Buddy #3127: Cancel object during printing with prusa connect

Twinsen68, 2023-07-06:
It will be nice to could cancel on objet during printing a sheet of multiple object. 
With beta firmware I have many failed whitch impose me to stop totally the print.

murk-sy, 2023-07-20:
Duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3073

Prusa-Support, 2023-10-25:
The new firmware [5.1.0-alpha2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha2) (or newer) allows this, however, it is still being tested and requires [PrusaSlicer 2.7.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1) (or newer).

Please allow me to close this issue as a duplicate.

Michele Moramarco
Prusa Research

closed 2023-10-25 by Prusa-Support

B-0-B-B-Y, 2023-12-10:
Is this actually a duplicate? This is requesting the ability to cancel an object using the Prusa Connect Web UI, not the physical printer's menu?