prusa3d/Prusa-Firmware-Buddy #3129: [BUG] MK4 keeps requesting filament path to be cleared upon dried spool change

bolsoncerrado, 2023-07-06:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**MK4**

**Printer firmware version** - [e.g. 4.0.5, ...]

4.5.4

**Original or Custom firmware** - [Original/Custom]

Original

**Optional upgrades** - [e.g. Filament Runout Sensor]

Stock Factory Assembled MK4

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint

USB

**Describe the bug**

1. Start a print
2. Spool runs out of filament
3. Filament sensor kicks in
4. Filament is removed BUT
5. even if you reply YES to the "was filament removed ok?" the printer KEEPS asking you to remove filament. Nothing helps, the dial, the clicky, nothing. Only a RESET, ruining the print.
6. Upon rebooting filament sensor works as intended, so there's no debris no filament messing the sensor, no nothing.

**How to reproduce**

Read above.

**Expected behavior**

Since it has happened to me already TWICE and non correlative, I asume this is a firmware glitch. I already blowed air, moved filament on the port hole, nothing helps when the bug is present. ONLY Reset button fix the issue and once the printer reboots, everything works as intended.

**G-code**
 
No particular gcode.

**Video**

Next time it happens I'll tape it.

bolsoncerrado, 2023-07-07:
@marcelismus what fw version did u have issues with? mine is 4.6.1 -- i'll install alpha3 now.

unclej84, 2023-07-13:
I am having the same issue with 4.7.1 firmware - goes to park position, unloads and shows confirm screen (selected "yes") but still shows to clear the path

unclej84, 2023-07-15:
Made a test with colorchange today and there the filament change worked - don't know what the difference between both cases (color change <-> filament runout) internally is

bolsoncerrado, 2023-07-15:
Idk either but I lost another print today again, pretty frustrating.

unclej84, 2023-07-15:
@bolsoncerrado are you on alpha3 now or still on 4.6.1?

bolsoncerrado, 2023-07-15:
Now on 4.7

arcenik86, 2023-07-16:
Confirmed issue on 4.7.1, however I did get it to clear after inserting and removing filament quickly a few times. Before doing so, verified there was no filament in the idler area. New MK4 build, third print.

bolsoncerrado, 2023-07-19:
I tried that technique and it didnt work for me :(

IftiBashir, 2023-07-24:
I would consider this a rather serious issue. Have Prusa responded?

bolsoncerrado, 2023-07-24:
No :(

zhiyao, 2023-07-30:
It happened to me as well twice. 

I realized the filament sensor is not calibrated for me and will always register as a detected filament.

Temp Fix: The only way to fix this is to do a control > filament sensor calibration.

Proposed Fix: There should have been a way to overwrite the sensor during such an issue.

unclej84, 2023-08-03:
Firmware 4.7.2 got released. Has anyone already tested, if it is fixed? Release note does not mention anything related.

bolsoncerrado, 2023-08-03:
Why would they? No one from Prusa ever commented here :(

MilleniumDawn, 2023-08-04:
Same issue on v5.0.0-alpha4. Stuck on requesting filament path to be cleared.
I'll give 4.7.2 a try...

derthomers, 2023-08-04:
same here on 5.0.0-alpha4. pushing in and pulling out a piece of filament multiple times worked in some cases. had it 8 times in 2 weeks. also on 5.0.0-alpha1..3. that really tends to ruin your print if you want to empty a spool completely before switching to a new one.

bolsoncerrado, 2023-08-04:
Happens also if you plan to leave and don't trust enough on the filament sensor stopping the print hence if you manually opt to change filament sensor may ruin the print awaiting you remove the filament you've already acknowledged to have removed correctly.

Pretty absurd altogheter.

arcenik86, 2023-08-04:
Just wanted to comment that this is easily duplicated with a filament change from the printer menu as well. Even after you confirm the filament is removed (yes on the prompt), you have to toggle the filament in and out for the printer to recognize nothing is loaded.

bolsoncerrado, 2023-08-04:
What I just said yep.

micro17, 2023-08-22:
Had the same odd behaviour today. Would be nice if Prusa can response if this is a software or hardware bug.

For me it cleared after putting the filament in and out multiple times

bolsoncerrado, 2023-08-22:
SOunds to me they have a severe hardware design issue with this filament sensor placement :(

This radio silence is nothing but worrysome!

I also wonder if XL people is having the same issue...

bolsoncerrado, 2023-08-23:
Well it's quite absurd the screen asks you "was the filament removed correctly?" just to ask you again to please remove the filament from the sensor path when you can't clearly reach it AND no other option (specially mid print) left than pray that wiggling multiple times the new filament will trigger the sensor :(

So I think it's both a hardware design flaw (like having a hole that gravity will end up filling with dust and such anyway) and a firmware issue...

FliesWithWind, 2023-08-26:
+1 on this. It just happened to me when I runed out of filament. Wasted 24h of print....

HaWiWe, 2023-08-28:
I had this happen a few times and could fix it by sticking the filament in and pulling out again a few times. It seems the hall sensor does not spring back, you can notice that there is no resistance when doing the above. At least that's what I think is happening.

unclej84, 2023-08-30:
In the Podcast they just mentioned that new final firmware with input shaping will be released (around) Friday - let's see if this included

Shushuda, 2023-09-06:
I just had this issue on 5.0.0-RC. The ball didn't spring back, there was no resistance. Inserting and pulling out a looot of times very rapidly finally made it spring back, resistance was there and the printer started detecting filament correctly.

bolsoncerrado, 2023-09-06:
I think all these posts prove they have a serious hw design issue...

Nulifier, 2023-09-14:
Something that worked for me to prevent the print from failing is to bypass the filament sensor when it gets into this stuck state.

What I did after receiving the remove the filament from the filament sensor message:

1. Remove the LoveBoard-cover-right using the M3 screw on the right side of the extruder. [(Link)](https://help.prusa3d.com/guide/5-nextruder-assembly_434014#435352)
2. Unplug the filament sensor cable from the lower connector. I found a pair of tweezers helpful. [(Link)](https://help.prusa3d.com/guide/5-nextruder-assembly_434014#435214)
3. The message should clear and you should be left with the normal load filament message
4. Install the filament sensor cable again
5. Reinstall the LoveBoard-cover-right (Its important to do it now as once you load the filament the extruder will start zipping around)
6. Load the filament

Obviously this involves some delicate work in the middle of a print and take your own precautions but it did allow me to resume a 6 hour print after hour 4. But definitely needs to be fixed on a hardware or software level.

bolsoncerrado, 2023-09-14:
Perhaps a cleaner, less risky way would be to add some small reset button somewhere to cut the power just enough for the message to clear....

bolsoncerrado, 2023-09-14:
Perhaps a savaged mechanical endstop? 😂

bolsoncerrado, 2023-09-14:
If anyone has links or names or part numbers to the male and female connectors may be I can order some and test a bypass system for this until a solution comes out. Granted this issue was opened on July 7th and no one from Prusa has chimmed in & granted we have had 2 firmware revisions in between i think it will still take a while for us to see a feasible solution and not just a temp fw patch...

Nulifier, 2023-09-14:
> If anyone has links or names or part numbers to the male and female connectors 

Pretty sure this is a [Molex 5025850370](https://www.digikey.com/en/products/detail/molex/5025850370/2819043)

bolsoncerrado, 2023-09-15:
Cool, what about the male One? :)

bolsoncerrado, 2023-09-15:
This? https://www.digikey.es/en/products/detail/molex/5025780300/2818999

Prusa-Support, 2023-09-15:
Thanks for reaching out to us.

This seems to be a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3056.
I'd rather close the first issue in favor of this one, as this duplicate grew in popularity.

We acknowledge the possible presence of a problem based on the number of reports on this issue.
However unfortunately we have not been able to confirm the presence of a bug so far.

We are working on this. If you encounter this issue again though, please follow these steps before reporting.

- Re-inserting the filament several times may knock off debris from the filament sensor mechanism.
- Find the cleaning hole on the top of the Nextruder, in the middle of it, next to the brass ring, and blow canned air through it.
- As the last resort, you can remove the "Nextruder sandwich" (brass ring + plastic parts, so you don't need to repeat the PG alignment) and remove the _filament sensor spring_ for deeper inspection and cleaning.

Also, make sure you have tested the most recent firmware version.
Consider testing the non-final versions available on [GitHub](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases) if you don't mind (could be unstable).

**Please refrain from disconnecting components or touching electronic components while the printer is powered on as this may cause damages.**

Michele Moramarco
Prusa Research

bolsoncerrado, 2023-09-15:
Wiggling constantly is the only thing that worked for me so far, as long  as I have patience and I don't move the print head accidentally during the process.... ::

bolsoncerrado, 2023-09-15:
And btw I think the "bug" is actually a hw design flaw as one of my 2 mk4 started failing almost "next day", so no many prints on it... :/

mataksik, 2023-09-16:
5.0.0RC same problem. Helped only disconecting filament sensor. 7h print was ruined

bolsoncerrado, 2023-09-20:
5.0.0 FINAL IS version: 

Not only I keep having the same old issue that now I have a NEW one too:

Printer requests to change filament when filament is still present and printing!! :( :/ :( 

[image]

enerbe, 2023-10-01:
I can reproduce it. Made a video. https://youtu.be/0cFO8x1V8tU

It is not even needed to be in the middle of a print. The video contains the process to reproduce.

bolsoncerrado, 2023-10-01:
Not the same last issue I reported here but yeah, pretty much no way out.... Unless you violently start moving up, down and dragging the filament on the sides until the sensor realizes you're moving it....

But having to reload filament midprint because the printer feels so is just meeeeh

ladislav37, 2023-10-05:
We have new MK4 (with 5.0.0 fw ver.) and have the same issue.. It looks like this is not a fw/sw problem but hardware... Most probably spring force is too low and it will not push the ball to correct place for filament detection. But this is only my assumption, because I can't find online any schematics or models for MK4 and I don't want to disassemble printer extruder right now.. :-)

bolsoncerrado, 2023-10-05:
Wdy mean schematics or models? To do what?

I guess you can take this as reference and perhaps source a stronger spring hmmm

https://help.prusa3d.com/guide/5-nextruder-assembly_434014

bolsoncerrado, 2023-10-05:
Plus the fact there's an opening on the extruder housing and any dust or buildup or even filament particles when pulling out the filament may collapse the system... IMHO the whole thing is flawed.

You may have issues if you put more pressure thru a new spring and the actual issue is ball does not pull back.

JonasFischer1, 2023-10-11:
I currently have the same issues on the MK4. During printing the filament sensor stops the print at random times and only allows printing again after a reset. 

The only thing that worked for me was to turn off the filament sensor in the settings menu.

eklynx, 2023-10-12:
I just ran in to this as well, changing filament mid-print due to spool running out, but it wouldn't recognize the filament was removed and i had to reset/abort.

bolsoncerrado, 2023-10-12:
Next time instead of reset  try inserting and removing and wiggling the filament quickly until sensor triggers...

eklynx, 2023-10-12:
Already tried that. Didn't help

unclej84, 2023-10-28:
@Prusa-Support Did I get the last Paragraph under "Current problems of MMU3 on the MK4" in the recent [blog post](https://blog.prusa3d.com/development-diary-update-on-mmu3-for-the-mk4_85043/) right, that you are also referring to this issue?

Prusa-Support, 2023-11-22:
The post on the blog kind of confirms these anomalies are most likely related to debris and irregularly shaped filament that may cause a mechanical blockage of the filament sensor moving parts.

As you probably already know now, they are working on improving the filament tip shaping through a sequence of movements. That is a work in progress but theoretically will drastically reduce the chances of filament sensing problems.

That mostly concerns Multi Material prints where several hundreds of filament changes may occur in a single print.
As of now, while we work on the best filament shaping, our best recommendation is to clean the filament sensor.
https://help.prusa3d.com/article/filament-sensor-mk4-mk3-9_465702
Nonetheless, sometimes reinserting the filament a few times can do the trick.

Michele Moramarco
Prusa Research

bolsoncerrado, 2023-11-23:
With all due respect these issues have been happening since day 0 way before mmu3 was even released :/

But ok. We'll take your word.

IftiBashir, 2023-11-23:
The issues have been happening since the beginning, but the seriousness has only been highlighted now since it causes huge issues when using the MMU3. Otherwise I wonder how long these issues would have been apparent before Prusa worked on a fix. In my opinion, this 'fix' with the change in filament tip should be implemented asap - MMU3 or no MMU3.

bolsoncerrado, 2023-11-23:
And TBH this filament tip thing is just a "patch" to a meh sensor configuration...

jakkaj, 2023-11-26:
Happening to me too now. Can't even get the calibration steps now after printer reset  :/ 5.0.1 firmware.

[image]
[image]

jakkaj, 2023-11-26:
The cleaning instructions here did help: https://help.prusa3d.com/article/filament-sensor-mk4-mk3-9_465702

jhenley01, 2023-12-03:
I've had this issue multiple times as well. Sometimes wiggling filament will fix it, sometimes not. Between this and the thermistor issues the MK4 has been disappointing. For the price of these machines compared to the competition you'd think they could get basic things like a filament runout sensor to work correctly.

PhLacoude, 2024-01-05:
I just experienced this issue for the first time after running out of PLA filament 1h20 minutes into a 1h51m print. This is VERY annoying when a simple GUI change could override the sensor.

bartbergman, 2024-01-24:
I had the same issue yesterday.
It's definitely a hardware issue (as already acknowledged by Prusa).
I had to manually clean the sensor because I didn't have compressed air available, but a tip from Prusa Support was to try to wiggle the metal ball inside the sensor with a magnet to unblock it. Haven't tried that yet, but will definitely try that if this occurs again.

n3wtron, 2024-02-09:
It happened also to me and I solved (that time) brutally disconnecting the printer from electricity, inserting the new filament in the extruder opening the "blocker", reconnecting the printer and finally clicking "change filament".

It worked, I don't know if it will always work but...

Prusa-Support, 2024-02-14:
The main cause is most likely debris around the filament sensor and this may affect pretty much all types of printers and filament sensors we tested so far, in various ways, and to different extents.
Of course, there may be specific cases where we can't exclude other causes of failure - check with Customer Support -  but again, it is usually a matter of debris. 

The first suggestion should probably be keeping the filament dry as moisture makes the filament more stringy and brittle.
Also, inspecting and cleaning the extruder gear and its surroundings is a simple practice that is often overlooked.
https://help.prusa3d.com/article/regular-printer-maintenance-mk4_419000

Disconnecting components while the printer is running is something to be _avoided_.
Unplugging the printer should not be necessary and may lead to complications. It may make the printer simply ignore the filament sensor status and encourage the customer to continue without solving the problem itself. I don't feel like recommending this method either.

Removing and inserting the filament a few times could knock off debris from the filament sensor, otherwise, you can clean the filament sensor with canned air, or take the sensor's ball holder off to proceed with cleaning.
https://help.prusa3d.com/article/filament-sensor-mk4-mk3-9_465702
Consider trying cleaning with compressed air _from below as well_ (through the idler door) but removing the ball holder for cleaning would be more effective.

Michele Moramarco
Prusa Research

pigskins65, 2024-02-14:
I have to respectfully disagree, at least with the firmware level I am at.
People have proven that they can get this issue to occur on demand. In my
case it happened every time I ran out of filament mid-print. I had to shut
off the sensor and pay close attention to the amount of filament remaining.

[76 more replies]