prusa3d/Prusa-Firmware-Buddy #3130: [BUG] Anti-aliasing for text labels on Home Screen icons gives dark verticals on some letters.

rubillos, 2023-07-08:
**Printer type** - MK4

**Printer firmware version** - 4.7.0

**Original or Custom firmware** - Original

**Describe the bug**
  For some of the letters in the font used for the labels on the Home Screen icons the anti-aliasing is causing some of the verticals to appear dark gray instead of white. Gives a very inconsistent appearance for the text. Occurs noticeably on 'h', 'n', and 'r'.

**How to reproduce**
  Look closely at the text below the Home Screen icons.

**Expected behavior**
  All of the letters should look solid.

Prusa-Support, 2023-09-27:
Thanks for reporting, however, I can't exclude a problem or a limitation at HW level.

Possibly provide pictures where the problem can be observed but first please reach out to the Customer Support for an evaluation of your printer hardware and electronics.

Michele Moramarco
Prusa Research

rubillos, 2023-09-27:
Pretty sure this is a bug. If you look at the file "src/gui/res/fnt_png/font_7x13.png" and look at the horizontal alignment of the different characters, the ones that have 'dim' vertical lines on the display are the ones which have pixels which start at "odd" horizontal offsets; the ones that look solid start on "even" offsets.

I think that what's happening is when the font png files are converted to hpp files there is a scaling operation taking place that combines 2x2 grids of pixels into a single grayscale pixel. For the characters that have pixels starting on even boundaries you get pairs of pixels combined into solid white pixels. For the characters that start on odd boundaries you get two "half" pixels that are gray.

rubillos, 2023-09-27:
If the font png files were adjusted so that all of the character's pixels started on "even" horizontal boundaries then all of the characters on the display would have proper "solid" vertical lines.

rubillos, 2023-09-27:
Did a bit more digging: looks like the font comes from "src/gui/res/fnt_src/LiberationMono-regular_9x16.png". Expanding that file and adding character box lines (for the first part of the file) gives:

[image]

You can see that the letters in question ('h', 'n', 'r', plus some others) have pixels that do not start at the left edge of the letter's box. I think this causes the anti-aliasing issue seen on the display.

Prusa-Support, 2023-11-08:
Interesting. Thanks for double-checking and explaining.
Our developers will look into this.

Michele Moramarco
Prusa Research

closed 2024-06-20 by github-actions[bot]