prusa3d/Prusa-Firmware-Buddy #3134: [QUESTION] Why Prusa Mini default probe offset

chezzandyto, 2023-07-11:
Hi prusa lovers,

I have a question.

Why Prusa Mini has a default 2mm offset between nozzle to probe ? and where modify it in code (for custom printer Buddy based)

I am triying to calibrate my printer (Firmware Buddy v4.4.1) with custom changes, but during test the nozzle is too close to the bed, i need to move up but the live z only allows go down. My pinda detects well the bed even more than 1.8mm. Prusa has an extra config to the nozzle to probe offset ?

Crusher7485, 2023-07-12:
If you need to move the nozzle up, then loosen the SuperPINDA probe, move it DOWN a little, tighten it, and re-do the calibration. Your nozzle will be higher now. No need to have the firmware changed in the code.

chezzandyto, 2023-07-13:
Of course, but my Pinda is too close to the bed.
Is possible to compensate in firmware ?

Crusher7485, 2023-07-13:
Wait, I'm confused. In the first post you said your nozzle was too close to the bed and you could not raise it with the Live-Z. Now you are saying your Pinda is too close to the bed. 

Which is too close, the nozzle or the pinda? They cannot both be too close, it is one or the other.

Prusa-Support, 2023-10-04:
Thanks for reaching out to us.

If you are close to the maximum adjustment of -2.000 then your PINDA probe is too low and kind of dangerously close to the nozzle height.
As a consequence, a lifted print would have higher chances of crashing onto the PINDA probe, a scenario we may easily avoid by setting the PINDA physical height "just right".

Theoretically, it may be safe to adjust very slightly beyond -2.000 but not recommended as in edge cases that might make the PINDA drag on the bed - a risk we would rather avoid.

Please move the PINDA probe "one ring" higher (roughly 1 mm) or less to be on the safe side!
Repeat the First Layer Calibration print starting from zero afterward.

I hope you find the answer satisfactory.
Please allow me to close this issue now.

Michele Moramarco
Prusa Research

closed 2023-10-04 by Prusa-Support

PhilippeGalid, 2024-06-24:
@Prusa-Support New Firmware is out 6.0.3 is in this Firmware the Display bug fixid ? Please some Information

Prusa-Support, 2024-09-21:
> @Prusa-Support New Firmware is out 6.0.3 is in this Firmware the Display bug fixid ? Please some Information

Not sure what bug you are talking about but it doesn't seem to be related to this issue in any way.

Michele Moramarco
Prusa Research