prusa3d/Prusa-Firmware-Buddy #3135: [BUG] PrusaLink /api/job response contains null nodes when print job completes

milstrata, 2023-07-12:
This issue is similar to #2941, but not quite the same.

**Printer type** - [MK4]

**Printer firmware version** - [4.7.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [None]

**USB drive or USB/Octoprint**
  USB drive

**Describe the bug**
While the printer is printing, making an http request to /api/job returns a result similar to the following. (I have reformatted the response for readability)
{
    "state":"Printing",
    "job":{
        "estimatedPrintTime":9295,
        "file":{
            "name":"filename.gcode",
            "path":"/usb/filename.GCO","display":"filename.gcode"
        }
    },
    "progress":{
        "printTimeLeft":960,
        "completion":0.88,
        "printTime":8335
    }
}

However, as soon as the printer finishes printing, the same http request returns nulls and it's not clear if the print job finished successfully or not:
{
    "state":"Operational",
    "job": null,
    "progress": null
}

**How to reproduce**
1. Get Prusalink password from printer's LCD (Settings->Network->Prusalink->Password)
2. Get Prusalink IP from printer's LCD (Settings->Network->IPv4 Address)
3. Select gcode file and begin printing
4. Make http request to /api/job. In bash, it can be done like so: curl --header "X-Api-Key: $api_key" http://$printer_ip/api/job
5. Wait for the print job to complete
6. Make another http request to /api/job
7. Observe that job and progress nodes in the json response are now null, where they previously contained data while the print job was running

**Expected behavior**
I expected the request to return data that shows the job is complete. I would expect the response to continue to show the job is complete until I select another file for printing, or start the print job over again. This would be consistent with the behavior of Octoprint's API.
Expected response would be something like this, where job->file->name is still present, progress->completion is 100, and progress->printTime is the total print job time in seconds.
{
    "state":"Operational",
    "job":{
        "estimatedPrintTime":9295,
        "file":{
            "name":"filename.gcode",
            "path":"/usb/filename.GCO","display":"filename.gcode"
        }
    },
    "progress":{
        "printTimeLeft":0,
        "completion":100.0,
        "printTime":9295
    }
}

**G-code**
The issue does not depend on the gcode file used.

**Crash dump file**
This is not a crash, but it is unexpected behavior of the /api/job http endpoint

**Video**
N/A

Prusa-Support, 2023-10-04:
Thanks for reporting.

Our developers will look into this.

Michele Moramarco
Prusa Research

samdurnxn, 2024-12-04:
Hello, can you tell me if this issue will be resolved ?
Additionally, the "status" value changes to 'operational' as soon as extruding is finished, but the printer is not actually operational until homing is complete. For example, with the MK4S and MMU3, the status changes, but the printer still needs to unload filament and move the head to the home position before being ready for another print.

danopernis, 2024-12-13:
Hi folks, this is quite an old issue. We are not working on this at the moment and there are currently no plans to work on this. My two cents from the developer's perspective:

1. It is difficult to actually tell when the print finishes, especially when using Octoprint. When feeding the printer with commands from serial line, you can always assume more commands will come and print basically never ends. Also, changing this can have some unforeseen effects on other parts of the system so I am a bit reluctant to change this.

2. To track job status in PrusaLink, printer would need to internally keep track of previous print jobs and start assigning some ids to them. This would require some additional memory and also would require to answer questions, such as when is the print job retired from the job list and whether it should survive printer reset. Storing the list to persistent storage also introduces potential for additional failure points.

closed 2025-04-29 by github-actions[bot]