prusa3d/Prusa-Firmware-Buddy #3140: [BUG] Nozzle cleaning failed error when using PC/PCCF

shledge, 2023-07-16:
**Printer type** - MK4

**Printer firmware version** - 4.7.1

**Original or Custom firmware** - [Original]

**Describe the bug**
When I use Prusament PCCF, sometimes nozzle cleaning repeats a full cycle (5-8 taps each cycle) on bed twice and then fails with a "nozzle cleaning failed" error.

If I reduce initial temps down to 210c in start gcode, its then less of an issue, but still is more inconsistent than compared to nozzle cleaning with PLA/PETG.

**How to reproduce**
Use any high temp material like PC/PCCF, let it do nozzle cleaning at its initial temp of 260c with bed at 110c. Used a satin bed, not sure if other bed types exhibit this behaviour.

**Expected behavior**
Should be able to continue after 3-4 taps regardless, or have the ability to continue printing anyway if nozzle cleaning fails.

pandaym3D, 2023-07-16:
I got the same problem, and I only use PLA.

shledge, 2023-07-27:
When I put it to 210c, the success rate goes up, but its not perfect and likely can cause bed levelling issues as any filament on nozzle is not hot enough to be squished out of the way.

Prusa-Support, 2023-11-16:
Thanks for reporting.

We were able to reproduce "_nozzle cleaning_" problems sometimes with PC-CF but not on a similarly enhanced and advanced material like PA11-CF.
This said, I have reason to believe that this is a matter of adjusting the temperature parameters in the "_start gcode_", which should be reported in [prusa3d/PrusaSlicer](https://github.com/prusa3d/PrusaSlicer) repository.

For the time being, letting the nozzle ooze for several seconds before the print, or withdrawing the filament 3-4 mm before the start of the print may ease the nozzle cleaning process.
Also, possibly make sure there is no residue of plastic around the nozzle.

Michele Moramarco
Prusa Research

closed 2024-06-20 by github-actions[bot]