prusa3d/Prusa-Firmware-Buddy #3141: [ENHANCEMENT] Summertime / Timezone

CommanderRedYT, 2023-07-17:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MK4
**Optional upgrades** - None

**Describe the expected functionality**
  A clear and concise description of expected functionality.

Currently the time menu only has an UTC offset, but this means one would have to change that everytime when time switches from summertime to Wintertime. This could be automatically done by including a list of timezones. Optionally one could find out the time zone via Public IP if the printer is connected to the Network and automatically set that

Prusa-Support, 2023-10-07:
Thanks for your suggestion.

This issue partially duplicates the existing issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/639 as an automatic timezone detection would remove the need for selecting timezones and virtually address this issue too.
However, that was an initial implementation of the feature on the Prusa MINI, so I'd keep both issues open for now.

Our developers will look into this.

Michele Moramarco
Prusa Research

Prusa-Support, 2024-04-04:
This issue is solved by FW [6.0.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0-RC1), which is a Release Candidate and therefore considered to be a rather stable pre-release.

Feel free to give it a go and share further feedback.

Michele Moramarco
Prusa Research

CommanderRedYT, 2024-04-13:
@Prusa-Support I do not consider this issue closed

Prusa-Support, 2024-04-13:
Not sure if the timezone can be detected automatically yet but the wintertime/summertime parameter should be implemented.

@CommanderRedYT Please check the description of PR https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3570 and test the even newer (and closer to a definitive release) FW [6.0.0-RC2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0-RC2).

Michele Moramarco
Prusa Research

closed 2024-06-15 by Prusa-Support