prusa3d/Prusa-Firmware-Buddy #3143: [FEATURE REQUEST] MK4 M117 Message on LCD Screen

oakbuffalo13676, 2023-07-17:
**Printer type** - MK4

**Is your feature request related to an existing problem? No

**Describe the expected functionality**
It would be nice to have the M117 message pop up on the LCD screen when I insert a pause through PrusaSlicer. Mine does not seem to do that to my limited awareness. Maybe it's just me, and I am missing something?

coaxial, 2023-07-20:
See #189

Prusa-Support, 2023-10-14:
Thanks for reporting.

I assume you are talking about the "[insert pause](https://help.prusa3d.com/article/insert-pause-or-custom-g-code-at-layer_120490)" (+ message) function of Prusa Slicer.

I'm afraid this feature was not fully implemented on the Buddy FW in the past and the new generation firmware for Buddy boards inherited the lack of support for the feature.
Our developers will work on it.

Michele Moramarco
Prusa Research

coaxial, 2024-06-20:
Isn't this stale only becasue Prusa doesn't work on it, rather than because it's been resolved?

jkotowicz123, 2024-06-23:
+1 Still need it in production environment running on Octoprint

coaxial, 2024-06-26:
Thank you for the information @Prusa-Support Michele. My printer is packed away in storage so I can't check out the new feature. Has anyone else here noticed the change? How is it?

Silvenga, 2024-09-09:
Still desired by every user. (100% of people polled, sample size is 1)

Prusa-Support, 2024-09-24:
> Other than this, Buddy Firmware is compatible with M117 as of now.
https://help.prusa3d.com/article/buddy-firmware-specific-g-code-commands_633112
The message is displayed in the place of the time estimates as soon as it gets triggered, so it works but please notice it would get overridden by the print time information after several seconds.

> Still desired by every user. (100% of people polled, sample size is 1)

Has any of these people had a chance to check this recently? 🙂 
It works as far as I can see so I wonder if this issue should actually be closed or am I missing something?

Michele Moramarco
Prusa Research

Silvenga, 2024-09-24:
> Has any of these people had a chance to check this recently?

I might be missing something, because I don't see it working at all when printing serially (it has always worked for gcode prints). I understand that I can navigate 3 levels deep into the printing UI to see a history of all the messages sent via M117, but that's not the same (that's what was called out in the firmware release notes recently).

This is what is expected:

[image]

Silvenga, 2024-09-24:
@Prusa-Support new issue, same as this original issue: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4220

I hope we can agree this isn't complete?

jkotowicz123, 2024-09-24:
+1 - As PrusaConnect is not ready for my production environment (no live monitoring, no plugins, not open source...) I still use Octoprint on all of my MK4s and I need to be able to display custom information on printer's display. To be honest, whole  UI while printing with Octoprint is way worse than it was on MK3S - you don't see any information from afar. Large, easily readable parameters and text of M117 in place of Octoprint's logo would be great start.

CZDanol, 2024-09-24:
Yes, not completed yet. Sorry, guys.

nkappler, 2024-09-26:
> Has any of these people had a chance to check this recently? 🙂
> It works as far as I can see so I wonder if this issue should actually be closed or am I missing something?

I can confirm this issue still exists on a MK4S
The text is shown for a brief moment, but as soon as the display cycles through other messages, like time remaining,
The message is gone forever.

Using a MK4S with stock settings.

CZDanol, 2024-12-19:
Internal ticket: BFW-6485

TimetravelerDD, 2025-06-09:
the issue is still very relevant. Showing a message on the screen is a pretty important feature

jim-roberts, 2025-06-10:
“The text is shown for a brief moment, but as soon as the display cycles through other messages, like time remaining,
The message is gone forever.
”

Yes this is a problem. I’m often not sitting right by the printer waiting for the pause. I can have prints with several pause events for inserting things like nuts at different layers and of different sizes. The message is a reminder of which type of nut is to be inserted. 

The feature on the MK3s waits for the user to acknowledge the pause before clearing.

CZDanol, 2025-06-11:
@coaxial not related to this issue, so I won't comment that here.

Sembazuru, 2025-09-15:
This does not work for me, but it might be because I serially print (OctoPrint). Should I open a separate issue about no messages when printing through the serial port?

3d-gussner, 2025-09-15:
@Sembazuru Thanks for testing. Yes it only works with USB Stick gcode. There is already an [issue about M117](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3653) not working on Octoprint.

Can you please upload your gcode to the USB drive and print from it to confirm that it is working?

jim-roberts, 2025-11-17:
Still an issue