prusa3d/Prusa-Firmware-Buddy #3148: Missing timmer

bentdragon, 2023-07-19:
### Description of the bug

I am trying to print a part with multiple colors, but when i parint there is no indication as to when the color is going to change??
Was there in the MK3. I have to sit and stare at it for 5 hrs to not miss a change? Is this a mistake or are we taking away good features for unknown reasons?

### Project file & How to reproduce

[BDFM_Knife_handle-2_0.4n_0.2mm_PLA_MK3S_6h0m.zip](https://github.com/prusa3d/PrusaSlicer/files/12097745/BDFM_Knife_handle-2_0.4n_0.2mm_PLA_MK3S_6h0m.zip)
[BDFM_Knife_handle-2.zip](https://github.com/prusa3d/PrusaSlicer/files/12097746/BDFM_Knife_handle-2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

ver 2.6.0-rc1+win64

### Operating system

Windows 10

### Printer model

Prusa MK4

rtyr, 2023-07-20:
I believe this is a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3107 and https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3138

Prusa-Support, 2023-10-04:
Thanks for reporting, however, this issue should be closed as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3107.

Michele Moramarco
Prusa Research

closed 2023-10-04 by Prusa-Support