prusa3d/Prusa-Firmware-Buddy #3149: [QUESTION] <Enter comprehensive title>

Skinnedknuckles49, 2023-07-19:
With a dual head factory assembled Prusa XL running firmware 4.7.1, when I connect a Rasberry Pi running Octoprint I received a Safety Warning which refers me to https://community.octoprint.org/t/octoprint-tells-me-that-my-printers-firmware-lacks-mandatory-safety-features-what-does-this-mean/350.

Basically they are saying "it means that it detected a firmware running on your printer that is known to lack or have broken implementations of mandatory safety features such as thermal runaway protection." It does mention that the firmware may have the protection and the implementation just be incompatible with their check.

Since I am confident that the XL implements thermal runaway protection, it would seem that Octoprint's test is not compatible with Prusa's firmware. I can disable this check in Octoprint but that l doesn't "feel right".

I tried to search the issues database and did not find the issue mentioned.

Skinnedknuckles49, 2023-10-12:
> Thanks for reaching out to us.
>
> I believe this is related to the issue #3137 <https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3137> .
> Yes, there is thermal protection but is not reported as it should.
> Our developers will look into this.
>
> About Octoprint itself, it is not directly supported by us and probably not much tested on a new printer model like the XL yet.
> Nonetheless, our developers plan to dedicate some time to the care of Octoprint and general serial line issues in the near future -
> #189 (comment) <https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/189#issuecomment-1731224779> .
>
> I hope this answers your questions.
> Can this issue be closed?
>
> Michele Moramarco
> Prusa Research
>
>
> This email has been checked for viruses by Avast antivirus software.
> www.avast.com

Thank you, this should close the issue.

Prusa-Support, 2023-11-23:
The issue will be closed.
Thanks for understanding.

P.S. You can find an update about the Octoprint compatibility adjustments here https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/189#issuecomment-1818998327.

Michele Moramarco
Prusa Research

closed 2023-11-23 by Prusa-Support