prusa3d/Prusa-Firmware-Buddy #3150: [BUG] Nozzle cleaning on M4 fails every time — but then continues

mix579, 2023-07-20:
**Printer type** - Mk4

**Printer firmware version** - 4.7.1

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
  PrusaConnect

**Describe the bug**
 Every time I try to print Prusament PC-CF, the "Nozzle Cleaning" fails. The error dialog shows "abort" or "retry". When I choose retry, it actually won't retry, simply continue with the print job. Feels like the wrong behavior but actually saves my bacon because every print completed just fine.

**How to reproduce**
Load PC-CF, start a print.

**Expected behavior**
1. The cleaning should succeed, not fail.
2. If failed, "retry" should—well—retry, not just continue
3. Add option for "continue", to ignore the failed cleaning because obviously it doesn't matter.

Prusa-Support, 2023-11-07:
Thanks for reporting.

Not sure why the printer would skip the new attempt and continue.
We will look into this but a video may actually help.

As for the main cause of the problem, we were able to reproduce "_nozzle cleaning_" problems sometimes with PC-CF but not on a similarly enhanced and advanced material like PA11-CF.
This said, I have reason to believe that this is a matter of adjusting the temperature parameters in the "_start gcode_", which should be reported in [prusa3d/PrusaSlicer](https://github.com/prusa3d/PrusaSlicer) repository.

For the time being, letting the nozzle ooze for several seconds before the print, or withdrawing the filament 3-4 mm before the start of the print may ease the nozzle cleaning process.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-11-16:
Allow me to close this issue as a duplicate of issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3140.

Michele Moramarco
Prusa Research

closed 2023-11-16 by Prusa-Support