prusa3d/Prusa-Firmware-Buddy #3151: [BUG] Print time not matching reality

Jothoreptile, 2023-07-21:
Prusa mk4 with latest alpha firmware.

Receive it monday 17/07 (assemble it in the night). Been printing with it since this time (but not 24/24). When a look at the Print time in the odometer I found the values really weird (like 3 day and 15 hours but I have been less than 3 days that I have it). And just after when lauching a 3 hours print the odometer shows a +6h on the print time (see photo just bellow).
[image]
[image]

nvtkaszpir, 2023-07-22:
only 3 days? 
here's better:

[image]

that's about 136 years.
:D

t-makaro, 2023-10-24:
I noticed this yesterday on my Mk4 with the 5.0.0 stable release firmware. 7 hours of printing (2 prints) and the time increased by 21 hrs.

amagreenhow, 2024-01-19:
I also have this problem on my MK4 with the 5.1.2 stable release firmware. The ODOMETER Print Time is currently increasing by about twice the actual print time for each print.
Today the ODOMETER Print Time went from 8 days 21 hours to 9 days 0 hours after printing a just over 1 hour 27 minute print and again after reprinting the same print it went from 9 days 0 hours to 9 days 3 hours

GlowingEye, 2024-03-29:
This issue still persists on my MK4 running 5.1.3 (stable release). The print time in the statistics is roughly doubled. I reset the printer twice, even reflashed the firmware and the issue is still there. In my example, the print took 10 min and 3 s while the statistics increased from 0 min to 25 min and 16 s (after reset, thereby zero before the print).

[image]
[image]

danopernis, 2024-06-20:
Although this issue is older than duplicate https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3220 I am closing this one because the title and description of the other one better _matches reality_.

closed 2024-06-20 by danopernis