prusa3d/Prusa-Firmware-Buddy #3152: [BUG] Missing newline in the output of `M46 M`

ondt, 2023-07-21:
**Printer type** 
MK4

**Printer firmware version** 
4.7.1

**Original or Custom firmware**
Original

**Optional upgrades**
Nothing

**USB drive or USB/Octoprint**
USB

**Describe the bug**
A newline is missing at the end of the output of `M46 M`.

**How to reproduce**
Send `M46 M` to the printer over USB serial. It returns this:

```
192.168.100.100
de:ad:be:ef:12:34ok
```

**Expected behavior**
It should return this:
```
192.168.100.100
de:ad:be:ef:12:34
ok
```

Note that `M46` (without `M`) works fine:
```
192.168.100.100
ok
```

**G-code**
`M46 M`

**Crash dump file**
Not needed

**Video**
Not needed

Prusa-Support, 2023-11-02:
Interesting.
Thanks for reporting, we are taking notes.

Michele Moramarco
Prusa Research

bkerler, 2023-12-18:
Fix has been merged, thus the issue can be closed.

Prusa-Support, 2024-01-17:
Sure. Thanks for the notice.

Michele Moramarco
Prusa Research

closed 2024-01-17 by Prusa-Support