prusa3d/Prusa-Firmware-Buddy #3153: [BUG] Incorrect information provided by `M115` (Firmware info)

ondt, 2023-07-21:
**Printer type**
MK4

**Printer firmware version**
4.7.1

**Original or Custom firmware**
Original

**Optional upgrades**
Nothing

**USB drive or USB/Octoprint**
USB

**Describe the bug**
The output of `M115` is incorrect on the MK4, it reports itself as a MINI, and it claims it has 5 extruders. Probably a copy-paste error.

**How to reproduce**
Send `M115` to the printer over USB serial. It returns this: (part of the output)
```
MACHINE_TYPE:Prusa-mini EXTRUDER_COUNT:5
```

**Expected behavior**
It should return this:
```
MACHINE_TYPE:Prusa-MK4 EXTRUDER_COUNT:1
``` 

**G-code**
`M115`

**Crash dump file**
Not needed

**Video**
Not needed

ondt, 2023-07-21:
For the purposes of automatic printer identification, is it safe to assume that a `Prusa-mini` with five extruders is actually an MK4? What is the number of extruders reported by the _actual_ MINI? Are these two values correct for other printers?

kozross, 2023-07-22:
This is what I get from my Mini: 

``MACHINE_TYPE:Prusa-mini EXTRUDER_COUNT:1``

DaHouzKat, 2023-10-28:
same here, i have the same problem with Firmware version 5.0.0

```
Send: M115
Recv: FIRMWARE_NAME:Prusa-Firmware-Buddy 5.0.0+12061 (Github) SOURCE_CODE_URL:https://github.com/prusa3d/Prusa-Firmware-Buddy PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa-mini EXTRUDER_COUNT:5 
```

bkerler, 2023-12-08:
Tested to work fine on MK4 with fw release 5.1
`FIRMWARE_NAME:Prusa-Firmware-Buddy 5.1.0 (Github) SOURCE_CODE_URL:https://github.com/prusa3d/Prusa-Firmware-Buddy PROTOCOL_VERSION:1.0 MACHINE_TYPE:MK4 EXTRUDER_COUNT:1`

ondt, 2023-12-30:
Thanks @bkerler, I can confirm that it's fixed on `5.1`.

closed 2023-12-30 by ondt