prusa3d/Prusa-Firmware-Buddy #3154: Prusa Mini dead Project

PhilippeGalid, 2023-07-22:
I would like to know if Prusa let the Prusa Mini die, because there have been no updates at all for a very long time, and no information about imput shaping and how the Mini will continue. And yes, imput shaping works very well on the Mini, there are enough Youtube videos with clippers that prove it and that it can print very quickly. I would like to stay with the original firmware, but if Prusa continues to let customers stick with the Mini and only be interested in the MK4 and XL, there is no other choice but to switch to Klipper. I ask for your support that we also get firmware updates for the Mini, I think enough of them were sold. Josef Prusa this is not OK

kozross, 2023-07-22:
From what I've heard, the Buddy board inside the Mini doesn't have enough RAM to get the input shaping the MK4 (alpha) has. However, I 100% echo your sentiment, considering that currently, the Mini lacks stuff even the MK3S+ has firmware-wise: I've had people _consistently_ surprised about this whenever I bring it up.

jonasniesner, 2023-07-23:
I fully agree that this is not ok. Promising software power panic and not delivering is another thing i dont like. And the missing gcode and screen command for Octoprint are also not ok. If you decide to lock the bootloader, you have to listen to the community and at least implement the features promised at release. The mini is a nice and reliable printer, but due to the lack of software support, i can not recommend it to other people anymore.

kozross, 2023-07-23:
I don't think it's even 'nice and reliable'. It skews constantly, the extruder jams at the first sign of trouble due to its heatbreak design, and its cooling is bad even by the standards of cheap printers, which is _really_ saying a lot. Plus, there's some _really_ strange corner-cutting going on in many places: heck, the LCD doesn't even have a back plate!

kozross, 2023-07-24:
@murk-sy - I _wish_ I could be as optimistic as you are. However, fact of the matter is, the Mini is trailing firmware-wise on things the MK3S+ has had since precisely forever, far before anything like the MK4 was even on anyone's radar. I mean seriously, it doesn't even prompt to eject on an M600, which the MK3S+ has done basically since forever!

coaxial, 2023-07-29:
The Mini is very clearly a second class citizen in Prusa's roadmap. 

There are serious issues open for 3+ years (lack of M600, lack of M73, filament sensor kinda sorta working) and nothing at all being done about it. 

The only hope is that because the MK4 is also affected and people are very unhappy that Octoprint isn't fully supported on the buddy board, this *might* finally make Prusa prioritize it. And then it will trickle down to the Mini. 

But yeah. Really not impressed. My next printer probably won't be a Prusa and I'm not recommending it to others either because of these issues and Prusa's complete contempt around them.

kozross, 2023-07-30:
@coaxial - I'm not sure this trickle-down will happen. The Buddy is not the BuddyX, not least of all due to having far less RAM.

coaxial, 2023-07-30:
I'm not sure either, just clinging at whatever hope I can... It has been very frustrating to see Prusa's contempt towards Mini users so far, especially those who want to use Octoprint.

tobimai, 2023-08-04:
Agree that it's sad. Especially when the Prusa team just claims that bugs aren't actually bugs, like #3022 where it drives into the end of the Z axis in some circumstances, which can't be good for the hardware.

matt-chapman, 2023-08-14:
I can't speak for everybody here, but I feel that I don't really care if it takes another year to see more movement on the MINI firmware, but I just want to see some _commitment_ to deliver features that were promised.

I would rather know what is going on than have to guess and fill in the gaps myself, while I watch the MINI build remain broken (you can't build from any tag newer than 4.4.1 right now) and have the feeling that my printer is getting left behind.

The silence is the worst - and I don't want to have to scrub through podcasts to hear what the plans (if any) are.

coaxial, 2023-08-14:
I echo the sentiment. I *do* care if it takes another year as we've already been waiting over three years at that point, but the silence and noncommittal answers are definitely a bigger problem. 

The Mini is still being sold, yet feels like a discontinued product, borderline pretending it never even existed. 

That's definitely not what I expected for a device sold as a "print farm workhorse", which implies (to me at least) full serial gcode support, full official accessory support (i.e. filament runout sensor), and not being met with either deafening silence or "feature, not bug" when reporting issues.

kozross, 2023-08-14:
I basically feel supremely disappointed. There are numerous, _mechanical_ issues with the Mini, some going back to its very release, which aren't even acknowledged. This goes well beyond firmware or poor decisions: the design is just a priori bad. Prusa not even acknowledging many problems exist, much less attempting to solve them, means that I no longer plan to buy anything from them, or indeed, suggest anyone else do the same.

It's sad, because I honestly thought Prusa were better than this, based on what I'd heard.

nicopasla, 2023-09-18:
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha1 will feel amazing pn the Mini, amazing news

matt-chapman, 2023-09-20:
> https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha1 will feel amazing pn the Mini, amazing news

I installed it on mine and ran off the included GCode samples, and boy does it _go_. Still need to spend some time using the IS profiles when slicing myself, but I am very impressed, and very grateful to see movement on the MINI firmware.

Prusa-Support, 2023-09-30:
`FW 5.1.0-alpha1` is the answer and closes this issue.
Please stay tuned for more.

Michele Moramarco
Prusa Research

closed 2023-09-30 by Prusa-Support