prusa3d/Prusa-Firmware-Buddy #3155: [BUG] MK4 Network UI shows LAN as `[stat]` instead of `[static]` when the menu is re-entered

gudnimg, 2023-07-22:
**Printer type** - MK4
**Printer firmware version** - 4.7.1
**Original or Custom firmware** - Original
**USB drive or USB/Octoprint** - N/A
**Describe the bug**
Missing character in the LAN type setting.

**How to reproduce**

1. Go to Network settings
2. Go to Wifi settings
3. Change LAN from DHCP to Static --> Note the UI shows `[static]`
5. Leave Wifi settings menu
6. Enter Wifi settings menu and observe UI bug --> UI shows `[stat]`

[image]

**Expected behavior**
  UI should show `[static]` always instead of `[stat]`

BastionNtB, 2023-09-24:
I'm seeing this on 5.0.0.

Also can't seem to get a static assignment to work either.

gudnimg, 2024-05-28:
Just checked using the latest MK4 firmware 6.0.1, the bug is still there:

[image]

gudnimg, 2024-08-19:
Likely still an issue, I will wait for FW 6.1 for MK4 and look into submitting a PR. The fix is trivial if I remember correctly.

danopernis, 2024-08-20:
@gudnimg yep, still there, even in 6.1.x series :slightly_frowning_face:

bkerler, 2024-09-01:
I just had a look at the bug isn't "trivial" to fix at first sight. It is related to the refresh of the WI_SWITCH_t display class as it seems : 
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/f5a498ab8d2a42341d0dbeb969b7ae047783e860/src/gui/MItem_network.cpp#L176
I assume that the length of "DHCP" is used being the first option in the list although the length of "static" should be used.

gudnimg, 2024-09-01:
If I recall correctly, then doing something similar as here:

```cpp
MI_NET_INTERFACE_t::MI_NET_INTERFACE_t()
    : WI_SWITCH_t(0, _(label), nullptr, is_enabled_t::yes, is_hidden_t::no, string_view_utf8::MakeCPUFLASH((const uint8_t *)str_eth), string_view_utf8::MakeCPUFLASH((const uint8_t *)str_wifi)) {
    if (netdev_get_active_id() == NETDEV_ESP_ID) {
        this->SetIndex(1);
    } else {
        this->SetIndex(0);
    }
}
```

was the trick. `SetIndex` would trigger a recalculation of the width. Its been a long time since I tested this to take it with a grain of salt.

bkerler, 2024-09-01:
Yes, that would be a valid "quick fix", but it doesn't seem to solve the actual real issue I think :)

bkerler, 2024-09-01:
I just wrote a PR. Not saying that I'm happy, but it does fix the issue.
```
MI_NET_IP::MI_NET_IP(NetDeviceID device_id)
    : WI_SWITCH_t(0, string_view_utf8::MakeCPUFLASH(label), nullptr, is_enabled_t::yes, is_hidden_t::no, string_view_utf8::MakeCPUFLASH(str_DHCP), string_view_utf8::MakeCPUFLASH(str_static))
    , device_id(device_id) //
{
    index = netdev_get_ip_obtained_type(this->device_id());
    if (index == NETDEV_STATIC) {
        this->SetIndex(1);
    } else {
        this->SetIndex(0);
    }
    // SetIndex doesn't update the displayed string length if idx==index, so we update it manually
    this->changeExtentionWidth();
}
```
Even though I expected that setting the Index accordingly (1 for static, 0 for dhcp), the internal index is already set and thus changeExtentionWidth isn't executed. By running it manually, it does update the string correctly. I will have a closer look why idx==index as this solves the issue but I'm not yet happy about having to manually run changeExtentionWidth to update the string length. Will discuss with the devs.

bkerler, 2024-09-01:
Ok, the bug went deeper as I feared but I was able to resolve it now. The problem is, that the MI_NET_IP was using a parameter "index" which wasn't defined locally but globally, setting as well the index value of WI_SWITCH_t .. which was ... really bad.

closed 2024-11-12 by github-actions[bot]