prusa3d/Prusa-Firmware-Buddy #3156: [BUG] Prusa Connect / Link Gcode Corruption on OTA gcode upload

jrgiacone, 2023-07-22:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4]

**Printer firmware version** - [5.0 Alpha 4]

**Original or Custom firmware** - [Original]

**USB drive or USB/Octoprint**
USB Drive formatted as FAT32

**Describe the bug**
When uploading gcode via prusa link and prusa connect over the air, gcode corrupts causing toolhead to move outside of path, sometimes it continues causing failed print, or the printer gets stuck and is unable to move.

For example the print head will move to the right or north of the object and pause/get stuck, print is unable to pause or stop.

**How to reproduce**
With alpha 4, upload a gcode file to the printer and check the file that was uploaded to the printer USB in gcode viewer and compare it to an export direct from prusa slicer.

Note if you download the file directly from connect cloud it is correct, the corruption is occurring when uploading the file to the USB over wifi.

**Expected behavior**
Gcode uploaded via wifi should be identical to a direct upload to USB via usb port

**G-code**
[Working and Problomatic Gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12137158/Working.and.Problomatic.Gcode.zip)

Note the files are the same gcode, 1 is direct from prusa slicer to usb, the other is via prusa link. Please compare the tool path in gcode viewer to see this issue. 

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

vorner, 2023-07-25:
I'd like some confirmation/clarification. By „Upload by wifi“ you mean the PrusaLink interface on the IP address of the printer / through IP address from slicer?

Does it happen every time, or just sometimes? How often?

Can you try with a different USB stick?

jrgiacone, 2023-07-25:
It has occurred via upload through IP from prusa slicer, direct gcode upload from prusa link, and file upload from prusa connect to the printer.

Last night did another one via prusa connect and it was identical so I don't think its on every upload. Have tried with 2 different usb sticks, the stock and a new one

JohnnyDeer, 2023-07-27:
@jvasileff we are trying to reproduce this problem and have a question to improve the reproduction process. Do you use MAC or Windows?

jvasileff, 2023-07-27:
@JohnnyDeer I was only analyzing the files posted by jrgiacone. I have not reproduced this myself. This issue caught my eye because I have seen Connect become out of sync with the files on the printer's USB drive, and Connect seems to trust filenames too much when a new version is uploaded without changing the filename, but so far that seems unrelated.

jrgiacone, 2023-07-27:
I have reproduced it on Windows and Linux, primarily using Linux. I don't have access to a mac. It does not appear to happen on every occurrence, I have found a power reset to sometimes fix it?

jrgiacone, 2023-07-28:
[gcode 2.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12189357/gcode.2.zip)
happened again, labeled the one from prusa slicer OTA upload

Prusa-Support, 2023-07-28:
Thank you for bringing this to our attention.

We acknowledge the presence of the bug and our developers are attentively investigating this issue.
Unfortunately, the bug is hardly reproducible though.

We would appreciate gathering further details for the reproducibility of the problem and possibly feedback from more users.

Here are examples of possibly useful pieces of information that we may want to collect.
- Exact steps and patterns followed by the users before the occurrence of the issue.
- Computer hardware and operating system information.
- Network and connectivity information like router model, distance of the printer from the router, internet speed tests made with other devices in the printer's position...
- Programs and operations running on the computer that may take up substantial computer and internet resources.
- Devices connected to the computer or to the same network that may take up substantial computer and internet resources.
- Number of devices connected to the same network.
- Status of the printer.
- Last operations performed on the printer.
- Third-party devices connected to the printer.
- Reproducibility of the problem via Ethernet or via a separate Wi-Fi network (e.g. mobile hotspot).
- Examples of corrupted files (before and after the data transfer).

Michele Moramarco
Prusa Research

jrgiacone, 2023-07-28:
Hi Michele,

- Exact steps and patterns followed by the users before the occurrence of the issue.

This has occurred both after a print completion has been sitting idle for a while, and even after a restart. Unsure if board temperature has anything to do with it, typically buddy board is 55-60C when printing.

- Computer hardware and operating system information.

Primary Computer is running Arch Linux kernel: 6.4.7
Intel i7 6700HQ
Prusa slicer: https://archlinux.org/packages/extra/x86_64/prusa-slicer/

Windows computer is running Windows 10 with AMD 5800x and Nvidia 3080. Running prusa slicer release 6.0

- Network and connectivity information like router model, distance of the printer from the router, internet speed tests made with other devices in the printer's position...

Modem: Spectrum Model EN2251
Router: Spectrum Model SAX1V1R
Distance: About 15 Feet away
Speed From Router is 450 Mbps Download and 20 upload from same position.

- Programs and operations running on the computer that may take up substantial computer and internet resources.

The only programs running would be Firefox and prusa slicer as the system was tested at pretty much idle.

- Devices connected to the computer or to the same network that may take up substantial computer and internet resources.

No Devices are connected to the laptop.
The desktop has 2 monitors, a mouse, and a keyboard.

- Number of devices connected to the same network.

Typically 6-10 Devices are connected.

- Status of the printer.

Unsure on this one, I think it typically Idle or ready to print?

- Last operations performed on the printer.

This has occurred both when transferring after a completed print, and after a restart.

- Third-party devices connected to the printer.

Nothing is connected to the printer.

- Reproducibility of the problem via Ethernet or via a separate Wi-Fi network (e.g. mobile hotspot).

Have not tried via ethernet, this has only been tested via wifi

- Examples of corrupted files (before and after the data transfer).
[gcode.2.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12195947/gcode.2.zip)
[Working.and.Problomatic.Gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12195948/Working.and.Problomatic.Gcode.zip)
Note files in the folder are labeled which was transfered over prusa slicer OTA (over the air) to printer.

When comparing the gcode 2 file, it appears to have randomly added in a filament swap that just moves the print head away and the back without any user interaction (not this was not intentional and was not present on the gcode uploaded directly to USB from the usb port on the computer).

I am happy to help test in any other was or provide any other information needed. @Prusa-Support

jrgiacone, 2023-07-29:
[Gcode Example 3.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12205362/Gcode.Example.3.zip)

It seems to have done it again uploading from prusa slicer OTA, note when I reuploded it there was no issues with the second one. Attached are screenshots of the differences in gcode
[image]
[image]
[image]

mix579, 2023-07-29:
If OP hadn't said s/he tried two USB sticks, this has all the hallmarks of a corrupted USB stick?!

jvasileff, 2023-07-29:
I took a closer look at `gcode 2.zip` and `Gcode Example 3.zip`, and what I found about the corruption is pretty interesting. Below, I'll refer to the "good" gcode file as `v1`, and the corrupted file as `v2`.

The corruption occurs in exactly 32-byte chunks. To analyze the corruption, we can first split each file into a bunch of 32-byte files with a command like `split -b 32 -a 4 v1.gcode`. Fingerprints for each chunk can then be created with `find v1 -type f -exec md5 {} + > v1-md5s`. We are then left with two files - `v1-md5s` and `v2-md5s`, which fully describe the original and corrupted gcode files in 32 byte chunks.

Analysis can be done with commands like `diff -u v1-md5s v2-md5s`, yielding:

```diff
--- v1-md5s     2023-07-29 12:09:55.952211454 -0400
+++ v2-md5s     2023-07-29 12:10:55.186357521 -0400
@@ -469,7 +469,7 @@
 MD5 (./xckhs) = 84e3ffd2863ba304f27d146a2dc59996
 MD5 (./xavya) = f3f08a555cef524001fab082ee6a6a50
 MD5 (./xakwl) = 285d1a7aa761de0db83219e19e28e388
-MD5 (./xakqu) = bed64c8ba35a13cbdbffcde3c9b39e27
+MD5 (./xakqu) = 862a0484fb481016590124d798f229f3
 MD5 (./xbwme) = 6bea35919a15b764ff5ed25351f4e07d
 MD5 (./xcand) = 75d0ae1e30c6b006b7ed840374c12b97
 MD5 (./xbhfs) = 48460aac9f132df8f4565c027a2cfad2
@@ -4472,7 +4472,7 @@
 MD5 (./xcikp) = 93bb8e33b593b70d49ab8b9c53c265c2
 MD5 (./xatzb) = 79eeb96b402d344ac274854716a61614
 MD5 (./xakwm) = dfd8a12e34b0080d2993659e5bf3e560
-MD5 (./xakqt) = 18c413f2d7af5294875aa3807c5e9cb7
+MD5 (./xakqt) = 3ef43351da57b392f9359baf6cc88d0e
 MD5 (./xcane) = 401827f26696585ff8b4e7199af159dd
 MD5 (./xbwmd) = 612723674e9ef948cb19986835f97226
 MD5 (./xbzqk) = 6804cafcfeab5fed4b980235d729a610
@@ -4887,7 +4887,7 @@
 MD5 (./xatze) = 715cfef9350c0c72eac8ffeb9e893f74
 MD5 (./xcikw) = 251d94b211b9fc8348990da3825b19f6
 MD5 (./xcimn) = 44c8ec41cdcf40294eda5bd97df537b7
-MD5 (./xakqs) = 081676dadbd1c9f40529aad093c99ae4
+MD5 (./xakqs) = d9bb7e5b7dc91f8b5c21112c0d128a41
 MD5 (./xakwj) = 5e9d00603a07adeb49d337beea986c44
 MD5 (./xcvfx) = ce05871851ab48e69b54fa18aaa6bdbb
 MD5 (./xbwkz) = 902a400864f977a95ee63a49475ba269
...
```

In red, we can see the hash that we *should* have for each chunk, and in green we see the hash that we *actually* have in the corrupted file. By looking at this diff, the following observations can be made:

* All corruption is indeed in 32-byte chunks
* The vast majority of "bad" chunks in the corrupted files exactly match content in the original file, it's just that the 32-bytes are repeated from some other location in the source file.
* The relatively few "bad" chunks in the corrupted file that cannot be found in the original file look like garbage or possibly FAT32 filesystem info
* *Interestingly*, "bad" chunks in the corrupted file often contain data "from the future" - that is, data that only occurs *later* in the original file. For example, in the first diff above, `862a0484fb481016590124d798f229f3` occurs in the corrupted file as the 472nd chunk. It also occurs in both the original file and corrupted files as the 20553th chunk. *This indicates that the corruption occurs sometime after the bytes have been sent over the wire, or that new bytes are overwriting previously written bytes.*

For `Gcode Example 3.zip`, there are 78 corrupted chunks. For all but 12 of those chunks, the data also occurs in the original file. The content of the other 12 chunks is:

```
% for i in $(cat chunks-not-in-v1); do echo "\n$i" ; cat $i ; done

v2/xbuzu
@!A!B!C!D!E!F!G!
v2/xbvac
?!?!?!?!?!?!?!?!
v2/xbvab
x!y!z!{!|!}!~!!
v2/xbvaa
p!q!r!s!t!u!v!w!
v2/xbuzw
P!Q!R!S!T!U!V!W!
v2/xbuzx
X!Y!Z![!\!]!^!_!
v2/xbuzy
`!a!b!c!d!e!f!g!
v2/xbuzt
8!9!:!;!<!=!>!?!
v2/xbuzz
h!i!j!k!l!m!n!o!
v2/xbuzs
0!1!2!3!4!5!6!7!
v2/xbvad
?!?!?!?!?!?!?!?!
v2/xbuzv
H!I!J!K!L!M!N!O!%
```

I have no idea what's going on here. But perhaps this will help someone with more knowledge of the PrusaLink firmware code or the FAT32 filesystem code. Regarding @mix579's comment, it does seem like a problem with the filesystem, but it's weird that the corruption is only happening when the files are being written by the MK4. Maybe using the USB stick on the Windows or Linux machine created some metadata that the MK4 can't handle?

rchiechi, 2023-08-12:
I am experiencing the same issue copying over wired LAN.
But I found that, with the example gcode attached (made from the example STL), I can trigger the bug by changing the filament profile to 3D fuel Pro PLA.

[example_gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12328056/example_gcode.zip)

[example_STL.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12328058/example_STL.zip)

I tested it back-and-forth twice each with the same positive/null result each time.
I do not know if it is triggered by switching filaments, generally, and I have had other prints with the 3D Fuel Pro PLA filament setting transfer correctly as well as prusament prints that broke. I only know that the steps outlined below were reproducible at least twice.

OS/Software:
Ubuntu 23.04
6.2.0-26-generic
PrusaSlicer-2.6.0+snap4

Printer:
MK4 kit with 5.0.0-alpha4 firmware
No third-party anything plugged in.

**Steps to reproduce:**

Open PrusaSlicer with default settings and load STL file.
Change filament profile to 3D Fuel Pro PLA.
Export gcode to disk from PrusaSlicer.
Export gcode to MK4 over LAN from PrusaSlicer.
Unplug USB stuck from MK4 and plug into computer.
Compare md5 sums of the two files and they will be different.

**Steps that do NOT trigger bug:**

*With PrusaSlicer still open to the same file*, change filament profile to Prusamment PLA.
Export gcode to disk from PrusaSlicer.
Export gcode to MK4 over LAN from PrusaSlicer.
Unplug USB stuck from MK4 and plug into computer.
Compare md5 sums of the two files and they will be identical.

antimix, 2023-08-13:
Hi Michele,
as programmer, even if I did not have time to look around the source code, the issue  seems related to a "rolling buffer", especially the _garbage from the _future__
Look for this occurrence:
- a pointer to a memory buffer variable (or to an array)
- an index pointer based on the offset of the buffer
- when bytes are received, they are added by the index pointer on the offset of the buffer.
- **unfortunately (check why) when the last chunk of bytes are received, they are written to the top of the buffer**
this can be due to: memory segment overlap, or incrementing and resetting the index pointer BEFORE writing the bytes in the buffer. This explains why _the bytes from the future_ that are the final part of the chunks, and they are written **on the top of the buffer** overwriting the correct one, and then written to the file.

jstm88, 2023-08-15:
I've also encountered the same bug on my own MK4 running (non-alpha) firmware 4.7.1.

I have noticed (and others have commented) how slow gcode transfers via PrusaLink seem to be, so there's definitely a bottleneck somewhere. If a circular buffer is in use, a bug like this would be possible in two places: when the buffer *fills*, or when the buffer *empties*. In other words, writing too much data *or* reading too far ahead could both be a problem. And since we have seen cases where the out-of-place block appears to come from the future, I would suspect that the USB write is the bottleneck and data from the network overruns the current index.

I'd be on the lookout for:
* An off-by-1 indexing error which allows the start/end of the buffer to overlap (and if this happens multiple times in a row, you'd see the "multiples of 32" issue)
* A race condition between the network writing to the buffer and the USB code reading from the buffer (in this case, multiples could be caused by timing)
* Possibly both?

I also think there needs to be some validation of the file after writing. A checksum would be the most logical approach. That clearly doesn't solve the problem, but it **would** make sure it was caught, and it would provide absolute certainty that a file was transferred correctly no matter what bugs may or may not exist.

antimix, 2023-08-15:
In the meantime, I  have just started to use my new MK4 with 4.7.2 and in order to check if this issue appears, I am using a FlashAir SD instead of the USB dongle (that is a USB Card reader with a FlashAir SD inserted ;) ) so that, since it is mapped as drive K: to my PC, I can immediate check with a compare app if the transferred file on K: is the same on my C: drive.
Till now on the 4.7.2 it never failed.

jstm88, 2023-08-15:
Just as a suggestion to others, here's what I'm going to do while this is being worked out:

* Export the gcode to a local file
* Upload via the PrusaLink web interface
* Download a copy from the web interface
* Hash both files and confirm a match
* Print

Of course, the *download* code may also have a similar flaw, so it will be interesting to see if this gives any false positives. But with this method, we can at least guarantee that there's no risk of damaging the printer, and we can still have the convenience of not having to shuttle USB drives back and forth (and put stress on the very tight USB port the MK4 has).

abjugard, 2023-08-16:
I haven't done the work to assess if I have experienced this issue myself, but from reading about it, @Prusa-Support if you're unsure where to start looking for problems, I'm gonna go ahead and guess this has to do with the frankly terrible network to USB file transfer code. Get a networking engineer to rewrite that for you and I bet the issue will disappear in the process.

vorner, 2023-08-16:
Reading through the comments in here again, I'll re-check one thing.

@jrgiacone Are you 100% you've managed to get a corrupt transfer _from Connect_ too?

My first thoughts at the time were pointing to certain newly introduced code (which'll admin I'm not really proud of, it was written a bit in desperation). The further comments here would actually strengthen the feeling. Except the code is _not being used at all_ in the Connect transfer, only in Link so I've ignored that feeling.

jrgiacone, 2023-08-16:
@vorner I believe I have, however, I've only tried connect a few times so I could have miss remembered, the majority of the issues have come from using prusa slicer to upload direct to prusa link through a physical printer.

With regards to connect I could be remembering wrong, I stopped using it because it would take way too long to upload any files.

jvasileff, 2023-08-17:
@jstm88 Have you had any failures? It seems that for those affected, the failures happen fairly often, but most people have no failures at all. It makes me wonder if there are different buddy board versions that run different code paths.

If corruption is reproduced with your script, it should be useful for troubleshooting as it could be tried in different environments.

jstm88, 2023-08-17:
@jvasileff I haven't noticed any more failures.

I actually ran several dozen prints on the MK4 after I built it without seeing any *obvious* problems. Those gcode files were all uploaded directly from PrusaSlicer, so the file on the printer would have been the only copy with nothing to validate against. It's possible there were corruptions but they weren't significant enough to show up in the prints.

I first noticed the problem on a relatively large one-off print and I saw a few of the external zits just near the top. I don't have the file any more but inspecting the print it definitely looks like they all happened towards the end of the file. The very next print I uploaded after that one had corruption throughout the entire file.

For what it's worth, I re-uploaded that specific file several times using my script and there was no corruption.

This actually has me wondering if the likelihood of corruption increases as the storage fills up. I'm using a 64GB Samsung drive so it would take a *long* time to fill it up, but I wonder if there might be some problem with higher block offsets, or potentially issues that happen as the allocation table grows. We'd need a lot more data to determine if that has anything to do with it. In other words, has this *ever* happened on an empty drive, and does it become more likely if there is more data on the drive? I can't say at this point.

rchiechi, 2023-08-17:
FWIW all of the corruptions that I encountered have been from a 256 GB drive that is nowhere near full. I also had fewer problems with smaller (in file size / print time) prints, where the corruptions is only noticeable as some small flaws in a couple of layers.

jvasileff, 2023-08-17:
> This actually has me wondering if the likelihood of corruption increases as the storage fills up. I'm using a 64GB Samsung drive so it would take a long time to fill it up, but I wonder if there [...]

Yeah, or if write sizes vary with fragmentation.

vorner, 2023-08-17:
There seems to be a lot of speculation in here… so, let me share something from what we have.

There are not different code paths depending on board version regarding the networking code.

We have a working theory (very much unconfirmed yet). If it is correct, the bug is very timing dependent. This involves both the speed of USB (which _does_ change with fragmentation, yes) and the „effective“ wifi speed. This seems to be the reason why some people can reproduce it often while others never see a corrupt file (we still haven't been able to reproduce despite bombarding printers with hundreds of transfers).

An interesting experiment could be to move the printer further away from the router, or put some interference between them, simply to see if the change of timing leads to the bug disappearing.

rchiechi, 2023-08-17:
> An interesting experiment could be to move the printer further away from the router, or put some interference between them, simply to see if the change of timing leads to the bug disappearing.

I get corrupted files transferring over a 1Gb wired Ethernet switch.
It is a 24 port Ubiquity switch (not a cheap no-name) and the printer and computer are each connected by less than two meters of cat-6 cable. 
No WiFi involved.

jstm88, 2023-08-17:
I can confirm my corrupted files also occurred with the printer connected via a 1Gb wired connection.

The upload speeds are *extremely* slow no matter what. The printer's USB implementation (either hardware or software) is the bottleneck. Local network transfers with other file servers are as fast as expected (near gigabit) and this also applied to OctoPrint when I was using it.

rchiechi, 2023-08-17:
> The upload speeds are _extremely_ slow. The printer is the bottleneck, although it's unclear if it's the networking code, USB code, or the USB hardware itself. Local network transfers with other file servers are as fast as expected (near gigabit) and this also applied to OctoPrint when I was using it.

Same for me: transfer speeds to the printer are far slower than they should be even for the 100Mb Ethernet interface on the printer. But I haven't tried OctoPrint, just PrusaSlicer and via Python requests.

vorner, 2023-08-18:
>  No WiFi involved.

OK, then our working theory is not the right one :-|

vorner, 2023-08-18:
Few more things I'd like to confirm:

* Each corruption is „unique“ ‒ that is, if the same file is transferred twice and both cases are broken, they are broken at different places (or sometimes not broken).
* This problem is specific for this version. The stable version (4.7.something) doesn't suffer from this bug. Have you also tried the newest one (5.0.0-RC, I assume it would still be present there)?
* Would anyone be able & willing to get a PCAP from the transfer? If we had a pair of PCAP and the corresponding broken file, we could try correlating them ‒ see if there are some interesting things happening around the time of the broken part (duplicate packets, retransmissions, …).

As for the speed of the transfer. I don't think I should be saying much before we have anything to show about it, but we are working on that one.

rchiechi, 2023-08-18:
@vorner Personally I have only had corrupted files on the 5.0.0-alpha4 when transferring from PrusaSlicer. I am now on the 5.0.0-RC and have been shuffling files to the printer by moving the thumb drive back-and-forth between the PC and the printer to avoid potential corruption. But I will try to find some time this weekend to reproduce the corruption. Do you want a packet capture of the whole transfer? If you have some tips for (I assume) tcpdump switches that would be great.

vorner, 2023-08-18:
Thank you.

Well, I want the part that breaks the gcode (if there's some), but I assume just sending the whole transfer / all communication with the printer during that time is much easier to provide. It shouldn't include anything private (the authentication to the printer uses digest-md5, so it should not leak the password, but if you're worried about that, it's possible to generate a new one in the menu).

I usually use something like `tcpdump -w printer.pcap host ip.of.the.printer` or wireshark with the same filter (`host ip.of.the.printer`).

jstm88, 2023-08-18:
@vorner
> * This problem is specific for this version. The stable version (4.7.something) doesn't suffer from this bug. Have you also tried the newest one (5.0.0-RC, I assume it would still be present there)?

I've been running 4.7.1 the whole time. I'll be updating to 4.7.2 soon (although the release notes don't indicate any networking/storage changes there).

antimix, 2023-08-18:
I am 4.7.2 (and the MK4 is connected via wireless) and I never had corrupted files.
So the issue should be specific for the 5.0.

jstm88, 2023-08-20:
Packet capture of the corruption happening in real time. Firmware 4.7.1.

I used my Python upload script for this:
```bash
❯ prusasend ~/MK4\ GCODE/RR\ Mini\ Revolver\ v1_0.4n_0.07mm_PETG_MK4_24m.gcode
Successful Upload!
SRC: 79a37fe30848bb60a237d350eb5c35a0f4a32922be549d6f2ec7924f48d78ec9
DST: 229a717298f1e7798606958eca9e5733922b2dff435ee80a590b5bb8fd70fe2c

UPLOADED FILE CORRUPTED
```

PCAP and Before/After GCODE files (the one with the timestamp is the corrupted one)
[mk4-upload-corruption.pcap.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12387028/mk4-upload-corruption.pcap.zip)
[gcodes.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12387027/gcodes.zip)

In the PCAP, there are *lots* of retransmissions and many are marked "TCP Window Full" by Wireshark. Should be plenty to prove that's where the corruption is happening.

Also you can see about halfway through the PCAP, the file finishes uploading and then the file begins being downloaded by my script. That download process proceeds without any issues, and the resulting hash is accurate to the file as stored on the drive. That confirms that downloading shouldn't be introducing any *further* corruption.

rchiechi, 2023-08-20:
@jstm88 I spent all day yesterday trying to trigger the bug on the 5.0.0-RC firmware and couldn't do it even after dozens of uploads.

I am curious if you also see glitches in the web interface. On the 5.0.0-alpha4 firmware (the one that produced corrupted gcode for me) the list of files on the web interface (on Firefox, Chrome, Edge and Safari / Linux and MacOS) would constantly refresh such that I could not scroll or click on any items. That problem is gone on the 5.0.0-RC firmware and apparently so is the gcode corruption---or least it has become so rare that I can no longer observe it.

mallek, 2023-08-20:
I am also getting this same behavior, I am using Wifi and within 3 meters of the Wifi router and the printer.  I have gotten the same behavior on 5.0 RC and I downgraded to 4.7.2.  I get this corruption on every upload via PrusaLink.

[PC_dragon_nice_1_0.4n_0.2mm_PLA_MK4_9h14m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12388967/PC_dragon_nice_1_0.4n_0.2mm_PLA_MK4_9h14m.zip)
[USB_dragon_nice_1_0.4n_0.2mm_PLA_MK4_9h14m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12388970/USB_dragon_nice_1_0.4n_0.2mm_PLA_MK4_9h14m.zip)

I can do a PCAP if you would like to help identify the error

rchiechi, 2023-08-20:
I hit the bug on 5.0.0-RC. Sorry I did not get a packet capture, but here is the source.gcode file and the broken dest.gcode file.

[broken in 5-RC gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12389244/broken.in.5-RC.gcode.zip)

jstm88, 2023-08-24:
So over the last few days I did some experiments, and I've found something very interesting. My key observation is that I have actually been *unable to reproduce the problem using the original, Prusa-supplied USB drive*.

The drive I have been using is a 64GB Samsung drive, FAT formatted. I tested the drive for integrity and it is perfectly fine. However, I tried filling both drives by repeatedly uploading files. The original drive would not show the problems even though it had a lot more data. That got me thinking, so what I did was do a bit-level clone of the 16GB Prusa-supplied drive onto the first 16GB of the Samsung drive, effectively copying everything I can about the formatting/layout of the original drive. Since then I have had no issues.

I have a couple theories:
1. The formatting on the 64GB drive may have some slight variation that the MK4 was not happy about. I haven't investigated this yet, but it's something to keep in mind. The FAT filesystem isn't particularly complex and I don't know what this may theoretically be. The formatting was done on a Mac, which *does* support formatting a 64GB drive as FAT. The documentation does say that Windows cannot do this, but it doesn't say it's not supported by the printer.
2. Leading to the other theory... that the printer simply does not support 64GB FAT volumes. Again, the Prusa documentation says that Windows and some OS X versions can't do it, but it suggests using a 3rd party tool, *not* that a >32GB drive can't be used.

I'll continue monitoring and see if the issue ever reappears with the reformatted "16GB" Samsung drive. It would be interesting to see if anyone has had these issues on a 16GB drive, and *especially* if anyone can confirm the issue on the Prusa-supplied drive.

Also: I updated my Python upload script to include statistics (so now you can be reminded just how slow the upload is!) and gave it a little better error handling. I also created a quick shell script to go along with it that selects a gcode file from a directory using `fzf` and sends it off, then archives the file *only if* it uploads successfully.

* `prusasend.py`: https://gist.github.com/jstm88/f25cffcd0c5a22e564fe8fe20114d3ad
* `psend.sh`: https://gist.github.com/jstm88/5cfc5726785e0a94554ddf19ce02293b

mallek, 2023-08-24:
I would like to add that I too was using a 128gb Samsung flash drive that I formatted the first 64GB in fat32

rchiechi, 2023-08-24:
I can't test the Prusa USB stick because it failed the first time I used it. I was trying to set up WiFi by creating a text file with credentials as per the instructions, but the text file became corrupted every time I plugged it back in to the printer. I tried formatting it, but it bricked, no longer recognized as storage. The printer also completely refused to print until I re-ran the initial calibration with a different USB stick plugged in to the printer.

Since then I've used a 256gb Samsung that I formatted with parted. It works perfectly if I copy files by hand. I only see corruption on network transfer, but, as noted, sporadically.

antimix, 2023-08-24:
Very Interesting discover @jstm88 !
I have just realized that I am using the FlashAir SD without problem, but is is 16GB, so may be that 32GB or less works well.
Nobody mentioned, but it ma be important, that when I cloned the files from PRUSA USB to the FlashAir (using Beyond Compare, file copying, not cloning the image) I noticed that there were a lot of hidden files and folders, like the USB has to be used on a sort of micro Linux system.
So I decided to copy also all that system and hidden files and folder over the SD, and since I had basically all the structure I had on the Prusa USB, it always worked with no issues.
- Did you all copy also all the hidden files & folders ?
If you just format the USB that structure is missing. So could be of value, use format FAT32 the 64GB USB, and then copy all the files structure, and see what happen. May be we identify if the issue is on the FAT format or on the files structure missing.

Crusher7485, 2023-08-25:
Just chiming in that I also don't think it's due to any hidden files or folders. My Prusa drive failed a couple weeks ago. I've been using this really cheap keychain USB drive that I normally carry on my keychain. It's super slow, only 8 GB, and I don't think I've ever formatted it myself. I had no issues with it, but I only used PrusaLink once or twice. I've used PrusaConnect a lot more, and the majority of the time I just transfer the USB back and forth because the printer is right next to my computer and if I have a large G-code I don't want to wait for it to transfer. Unless my printer is already printing, then I'll use PrusaConnect to transfer newly sliced files as it prints since I can't unplug the USB drive. 

I don't think I have a USB drive bigger than 32 GB or I'd try it and see if I too got corruption on a large drive but not a smaller drive. I do have a 2 TB SSD, but it's formatted with Linux Mint as a bootable OS and I'm not sure I want to bother moving partitions around just to use it as a bug test device.

jvasileff, 2023-08-25:
FWIW, I've been using a macOS formatted SanDisk 128GB usb drive without any problems.

While anything is possible, I'm a bit suspicious that disk size matters, since 1) I assume well tested third party code is used to read and write the FAT32 filesystem, and 2) the corruption happens in small chunks, but my hunch is that overflow bugs in the filesystem code would produce more significant corruption.

That being said, one test might be to manually partition to force the start of the partition to be towards the end of the drive, to exercise the code that writes to the block device. Another test would be to partition normally, but then mostly fill the disk (on a pc) with large files, hoping to push further writes towards the end of the filesystem.

Aside from size, it seems like the state of the filesystem and USB drive might matter, since that might affect timing and what happens in the code that reads from the network and writes to the drive. Testing with an old/cheap usb drive or SD card with a USB adapter might be useful, as would testing with a heavily fragmented filesystem.

Time permitting, I may try some of these things next week.

sjors-lemniscap, 2023-08-26:
Experiencing the same issue here. After hours of troubleshooting why one layer of a particular print kept failing I figured that my gcode, uploaded through WiFi, was corrupt.

My setup:

- Prusa MK4
- Firmware 5.0.0-RC
- Samsung Fit 256GB ‎MUF-256AB/APC
- USB formatted as FAT32 on MacOS Ventura 13.4.1 (c) with Master Boot Record partition scheme

Thanks to [jstm88](https://github.com/jstm88)'s python script I continued uploading over WiFi and discovered every now and then a file to be corrupted, after deleting the file from the printer and uploading it again it usually validated successfully.

The interesting part is that one particular file (attached the gcode and stl for reference) [sheet-holder-hex.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12445431/sheet-holder-hex.zip) kept getting corrupted, no matter what I tried (renaming the file, generating the gcode again etc.). I decided to reformat my Samsung USB drive and after the first try it validated and processed the file successfully.

Considering that; the Samsung Fit 256GB MUF-256AB/APC is a relatively fast USB drive, and others that are using a relatively new Samsung USB drive seem to experience similar behaviour, perhaps the issue could be that issue might be with the transfer speed of the USB drive. Just wanted to share my thoughts and experience here. I will purchase / try another (slower) USB drive to see if that resolves the issue for now.

rchiechi, 2023-08-26:
> Experiencing the same issue here. After hours of troubleshooting why one layer of a particular print kept failing I figured that my gcode, uploaded through WiFi, was corrupt.
> 
> My setup:
> 
>     * Prusa MK4
> 
>     * Firmware 5.0.0-RC
> 
>     * Samsung Fit 256GB ‎MUF-256AB/APC
> 
>     * USB formatted as FAT32 on MacOS Ventura 13.4.1 (c) with Master Boot Record partition scheme
> 
>     * Printer connected over WiFi, Access Point is on the other side of the wall (facing backwards) of where the printer is positioned
> 
> 
> The interesting part is that one particular file (attached the gcode and stl for reference) [sheet-holder-hex.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12445431/sheet-holder-hex.zip) kept getting corrupted, no matter what I tried (renaming the file, generating the gcode again etc.). I decided to reformat my Samsung USB drive and after the first try it validated and processed the file successfully.

I have had *exactly* this experience and am using identical hardware/firmware, but with printer on Ethernet. I even had [a gcode that reliably triggered the bug](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3156#issuecomment-1692365680) until reformatting the USB drive!

sjors-lemniscap, 2023-08-27:
> I have had _exactly_ this experience and am using identical hardware/firmware, but with printer on Ethernet. I even had [a gcode that reliably triggered the [...]

Interesting.. long story short I **feel** like (and as mentioned before by @jstm88) this is rather an issue of how the Prusa buddy board of the mk4 reads from the buffer and writes it to the USB, where it looks like that a fast USB drive is causing a race condition... Even though it still wouldn't explain why after formatting the exact same USB drive the issue didn't occur (so far). I'm curious whether the issue will slowly start to re-occur again the more I write to the USB drive.

We have at this point pretty much ruled out that the issue is specific to the WiFi network stack as it seems to occur on the LAN network stack as well.

Last but not least; a friend of mine has also a Prusa MK4 and uses a slower, Kingston, USB Drive (will edit later with the exact model number). He compared about 40 of his gcode files uploaded through LAN and WiFi to this Kingston USB Drive with the source files on his computer and has **0 corrupted files**.

rchiechi, 2023-08-27:
> Last but not least; a friend of mine owns also an Prusa MK4 and uses a slower, Kingston, USB Drive (will edit later with the exact model number). He compared about 40 of his gcode files uploaded through LAN and WiFi to this Kingston USB Drive with the source files on his computer and has **0 corrupted files**.

I did something similar by automating the upload of a few files dozens of times for several hours. Over dozens of uploads *of the same selection of gcode files* I got zero errors. But with specific gcode files I got the error on every upload... until I reformatted the USB drive.

In my hands, two things seem to be true: 1) the bug isn't triggered until the USB drive has been in use for a while (i.e., had files written to it / printed) and 2) once a file triggers the bug, it can (but not always) do so reliably. But it really hard to pin down the variables that affect the bug without having a dedicated printer with several different USB drives, etc. to test.

vorner, 2023-08-28:
Reading through the bunch of new posts and especially the interesting findings about reformatting the flash drives. USB drives use relocation tables (tables mapping virtual block numbers to physical ones, to balance the load of the memory and to stop using broken blocks). Formatting uses a `TRIM` command which effectively throws the current table away and starts a fresh (empty) one. Which leads to several things:
* Any potential „glitchy“ block (one that may still hold the data, but needs more tries to write into, or is slow or…) is either discarded or moved somewhere else.
* Timing changes (both because now the drive needs to scan smaller relocation table and because different blocks are now used).

It also seems that the bug was „always present“ (cases of corruption on 4.7.1 reported above), but maybe because some timing issues it is now much more probable to happen.

Yes, we use third-party code for FAT & USB, but maybe it's time to start investigating these too.

Side not for the issue of cold nozzle and stuck printer. It very much sounds like the printer is desperately retrying reading data from the USB drive and not getting it.

Cbutters2000, 2023-08-29:
[BADGCODE2.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12468976/BADGCODE2.zip)
I'm also experiencing this again, with a SAMSUNG 64GB USB 3.0 drive
Using PrusaSlicer, exported gcode to the PC, then copied over PRUSALINK to the Printer. Had a bad print. Compared the two SHA1 of the files on the PC and the one on the USB and they are different.

michalmiddleton, 2023-08-29:
I also have this issue - I use Samsung USB drive (SAMSUNG MUF-64AB/AM FIT Plus 64GB). The bug exists for me on 4.7.1 and 4.7.2. (I haven't tried any other firmware releases). 
Formatted to FAT32 with MBR on Mac. Printing via PrusaLink & wifi.

DRracer, 2023-08-31:
Dear reporters, our dev team spent a considerable amount of time trying to reproduce similar issues like you are reporting. So far we haven't been able to reproduce blocks of G-code randomly copied to various places, but in one case we have seen a block missing. 

Currently, we ran out out of ideas. We suspect some data-race or timing issue somewhere around the filesystem/USB  level, probably related to MK4+USB drive combination (we don't know that yet).

**Therefore, we'd like to ask for your help.** 

1. Is there anyone who is currently **repeatedly and reproducibly** experiencing the issue (every second or third corrupted G-code is reproducible enough) and willing to share their machine + the unformatted USB flash drive? Please report either directly to this thread or to my github email.
We offer a replacement MK4 + another USB drive + something more in reward.

2. Also, if there is anyone willing to share a `dd` image of their USB flash partition, that may help us as well. We don't know if it's just the filesystem level or if some USB communication timing is required to trigger the issue.

Thank you, good luck :crossed_fingers:

antimix, 2023-08-31:
Hi Developers,
I got another idea, but you have to check how it applies to the reality since I don't know too many details of the HW.

**FACTS:**
- Lots of people had problems with the LCD display that blocked or messed up the LCD images
- A possible culprit was addressed to the RIBBON CABLE whose data signals were disturbed
- Some others addressed the culprit only to the second LCD board version.
-  My LCD never had problems, and I also never had problems of file corrupted.
-  I have the LCD board version that is supposed to work (the first one)
- I put extreme attention to perform the LCD ribbon cabling keeping away from areas subjected to possible disturbs

**SUGGESTIONS:**
The file system logic is handled by the main Buddy board that communicate with the LCD (that contains the USB dongle) and so it sends the data back and forth to the LCD board through the ribbon cable.

- **is this issue happening to people that had LCD issues in the past ?**
- **What board version have they ?**
- May be that the problem is due to data signal error on the ribbon, and while a wrong pixel is difficult to detect on the LCD, a wrong byte in the FAT or file under checksum is more noticeable and do the mess.
- I personally had no issue at all on my LCD of my MK4 kit, and I put extreme care on the LCD cabling.
- may be people who have file corruption suffers also of LCD signal corruption.

Cbutters2000, 2023-08-31:
> Dear reporters, our dev team spent a considerable amount of time trying to reproduce similar issues like you are reporting. So far we haven't been able to reproduce blocks of G-code randomly copied [...]

My machine is doing this regularly. I would be willing to volunteer my machine+USB Stick for your testing. Please message me and I'll also email you at your email.

antimix, 2023-08-31:
As example I am using a FlashAir (32GB) SD on a USB Dongle Card Reader, and no issue.
So the space is 32GB and the card reader can "slow down" the data rate.

however I am between:

1)  40% - FAT32 library bug  for big disk sizes
2)  40% - Data signal error on ribbon cable between the Buddy and the LCD board
3)  20% - Protocol error for FAST data rate USB dongles.

the fact that many users with **constant** errors on a file, solved the issue reformatting the USB or changing to a smaller size make me think more for option 1.

DRracer, 2023-09-01:
> Komu: prusa3d/Prusa-Firmware-Buddy ***@***.***>
> Corruption on OTA gcode upload (Issue #3156)
> "

Hello Antimix,

  your mind is heading the similar way like ours, what you wrote are valid 
comments and recommendation ;) . We are working on it real hard, trust me, 
it's a serious issue which can cause all kinds of unexplainable behavior. We
also suspect some kind of HW+SW related corruption. 

However, since we haven't been able to reproduce the issue in any similar 
way, the next step is to collect a few customer's printes and investigate 
them.

Thanks for reaching out to me,

best regards,

Ing. David Rehor

FW DEV team leader at Prusa Research

---------- Původní e-mail ----------

   

   

(https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3156#issuecomment-1700922085)
"

DRracer, 2023-09-01:
@Cbutters2000 Thank you for volunteering your machine for the test and your email. Our tech support team should reach to you shortly to organize the exchange.

abjugard, 2023-09-01:
> @DRracer It might be helpful to post that on the Prusa forums, and maybe even on Reddit - not everyone is following along here. From what I see so far the only common thing is the usage of USB drives with 64GB or more (AFAICS everyone affected reported that they are using such a large drive). So it might be that writing to higher block numbers causes that problem (when the drive is getting fuller).

Honestly it isn't all that surprising that people are replacing the USB drive, since the drive that ships with the kit is super slow, people would rightfully be extrapolating that slow transfers might be because of a slow drive, that's at least the sole reason I've installed a faster and larger drive (128Gb version of [this one](https://www.westerndigital.com/products/usb-flash-drives/sandisk-ultra-dual-drive-go-usb-3-1-type-c#SDDDC3-128G-G46)) in my printer.

If the problem stems in filesystems formatted in a way the printer doesn't like, why not add a feature to allow the printer to do the formatting?

I'd much rather the printer got better at working with filesystems created elsewhere (including exFAT please, or just anything that supports modern metadata), but this would be a good stopgap solution while we find the real problem.

DRracer, 2023-09-05:
@abjugard actually the slow USB read/write is caused by something else than a flash drive by itself, we are working on a solution to boost the transfer speeds.

Formatting a flash drive in the printer doesn't make too much sense, it uses standard FAT32 - it either works on doesn't. What we are trying to find out in this issue is how it is possible, that the data get corrupted.

And yes, the "unable to reproduce" tag is still perfectly valid on this ticket, because we really haven't been able to reproduce it, even though there are people experiencing the issue (and we are listening to them therefore machines are being collected to reproduce the issue).

@wizbongre That sounds suspicious, but - the 5.0.0-RC had some issues with upload speed, but **keep your :eye: on it - if you cannot reproduce the issue with this specific version, please report back**. It would be an important piece into the puzzle.

wizbongre, 2023-09-05:
@DRracer to be 100% clear - I originally experienced the issues on 5.0 RC, so rolled back to 5.0 Alpha. I still had issues.
Now that I've updated to 5.0 RC with a forced reset, the problem appears to have disappeared.
I can't find a way to check the version of the WiFi module - is that possible, so maybe others on here could compare and see if there is any link?

vorner, 2023-09-08:
I'd guess the wifi isn't the cause of the problem. For one, the ESP firmware is tied to specific version by the printer (will upgrade or downgrade as necessary, and it is checked on every boot).

For another, there are people reporting the problem on ethernet, with no wifi involved.

Furthermore, looking at the corruptions posted here, the „copy-pasted“ chunk of gcode is sometimes as far as some ~36kB (and always multiple of 4096 bytes) away and we are pretty confident we don't have a buffer this huge anywhere near the network stack. Both the 4096 and 64B mentioned above (which are offsets of corruptions from beginning, and lengths of the corruptions) suggest it is somewhere near the USB rather than network.

The bug seems to be very fragile in the sense some (unknown) conditions must be met for it to exhibit ‒ that is, wifi might be triggering it for some (maybe due to different timing). Unfortunately, that means that the bug seemingly „disappearing“ is more likely just disturbing the trigger than actually „fixing“ it :-(.

[47 more replies]