prusa3d/Prusa-Firmware-Buddy #3161: [ENHANCEMENT] Redesign PrusaLink login/password credentials and interface

neuron-whisperer, 2023-07-24:
**Printer type** - MK4 built from kit

**Optional upgrades** - None

**Is your suggested improvement related to an existing problem? Please describe.**

Currently, PrusaLink is accessible via username/password. The username is hard-coded as "maker" and password is a randomly generated, fixed-length alphanumeric string that is difficult to remember and type. The UI allows the user to re-roll the password, but does not allow a user to set a preferred password. (Obviously, it is desirable to allow the user to specify both username and password, but that is not really the point of this submission.)

Most modern web applications that require username/password credentials are coded to embed username and password fields in a login web page. Those fields are tagged with metadata that allows password managers (like 1Password) to recognize them, to record credentials when they are first entered, and to inject them for subsequent login. Password managers vastly improve both user convenience (due to not having to remember and type passwords for websites) and security (due to promoting unique passwords for each site, in lieu of reusing passwords across multiple sites).

**Problem:** The MK4 web interface does not request credentials this way. Rather, the MK4 web interface requests these credentials via Basic HTTP Authentication. Most browsers (including Google Chrome and Apple Safari for iOS) handle Basic HTTP Authentication by popping up a separate window or tab with Username and Password fields, and a "Submit" button that transmits entered credentials to the web interface. Unfortunately, Basic HTTP Auth is not compatible with password managers like 1Password - it can neither record entered credentials nor inject them for subsequent login. Because of this design choice, users must manually type in (or, at best, copy-and-paste) both the "maker" username and the randomly-selected alphanumeric password.

This problem is not only a user convenience issue - it is also a security issue: forcing the "maker" username for all user accounts, and forcing all passwords to conform to a fixed-length alphanumeric format, vastly reduce the entropy of user credentials and make intrusion via brute-force password-guessing much more feasible. An attacker who succeeds in brute-force guessing the password could interrupt a print, inject malware that invades the user's LAN, or **even start a fire** by misconfiguring the heating elements of the MK4.

**Describe the expected functionality**

(1) PrusaLink must allow users to choose their own username (instead of forcing everyone to use "maker") and also allow users to specify their own passwords (instead of forcing everyone to use randomly generated, fixed-length alphanumeric passwords).

(2) PrusaLink must encode the username and password fields in a login web page like most modern web applications, instead of using Basic HTTP Authentication, which is not compatible with password managers.

mix579, 2023-07-24:
Has been proposed before but is as valid as it ever was.

bolsoncerrado, 2023-07-27:
I second this!

+1 please!

ondratu, 2023-07-27:
I just correct some mistakes in text.

* PrusaLink use HTTP Digest, not HTTP Basic Auth, the Digest is more secure
* HTTP Digest is standard way, it exist RFC: https://www.rfc-editor.org/rfc/rfc7616
* Difference between credentials filled to HTML forms and to this extra browser window is just that HTTP Digest does not sent password in plain text to printer, only special hash generated in defined (in RFC) way. This is ok for TLS connection, which Buddy firmware can't do it.
* That means that LAN malware can get password from HTML Form much more easier.

More secure ways are:
* switch to SHA instead of MD5, which is used in HTTP Digest, but only Firefox can do that.
* switch to TLS connection which is hard to manage on LAN devices and hard to implement to Buddy FW.

aderusha, 2023-07-29:
Here's another suggestion: display a QR code with the credentials and the URL of the machine. Being able to do so is another advantage to digest auth!  That way the user can scan the code and be brought to the web page already logged in.

eg: `http://maker:abcdefg12345678@192.168.1.10`

bolsoncerrado, 2023-07-29:
Thats nice if you want to quick login with a phone or similar but for a centralized operation like a Farm environment....

aderusha, 2023-07-29:
It's just a way to get you in the first time so you can set a password, but you could also save the link generated to take to your desktop.  Would work fine for farms too!

nullbreached, 2023-08-18:
If only for compatibility with password managers! (And hopefully simultaneously fixing issue with Safari [https://github.com/prusa3d/Prusa-Link-Web/issues/386](url)

Liniik, 2023-10-21:
I second this, when you're doing something like that, why would you ever do that like it is now? More so considering you can't even change the bloody password... To say it's inconvenient is understatement - it's the one and main reason why I don't use the web interface!

bolsoncerrado, 2023-10-21:
With the XL debacle and the lateness with their "web admin" thing "competing" with Octoprint they're getting soooo late to the party that I foresee a dark future for this company, VERY sadly! :'(

atollis, 2023-11-04:
Yeah this web interface is very half baked...

bcnine, 2023-11-13:
Is this for real?! I can't believe there's no way to set the f'ing password! I love this printer, but COME ON Prusa! This is just embarrassing!

QueDuck, 2023-11-21:
Shouldn't it be up to the business or individual as to whether they want this level of security. Secure login with complex password or none at all?  Also how often have we seen passwords written on a sticky note stuck the monitor or on the underside of the keyboard.

rkrussell, 2023-12-28:
I completely agree with the OP. This is unnecessarily badly implemented.

JeremyDWilliams, 2024-01-05:
As above, QR code to link to website (IP) automatically, ability to turn off password and at the very least ability to edit user name AND password. @neuron-whisperer added label of enhancement, which it would be, but honestly this is more than that. Safety, security and usability are issues that perhaps rank above a mere enhancement imo :)

Liniik, 2024-01-05:
> As above, QR code to link to website (IP) automatically, ability to turn off password and at the very least ability to edit user name AND password. @neuron-whisperer added label of enhancement, [...]

TBH the things you mentioned, except the QR code, are not an enhancement but they are BASIC FEATURE so currently it's a bug... I really can't believe that there's this thread and we still don't have a reaction from a Prusa dev half a year later!

TorgeirHolm, 2024-01-21:
Completely agree! Please change this to make logging in to PrusaLink smoother.

twshield, 2024-01-22:
This should be closed, v5.1.2 uses digest authentication with a username and password.

Note that this means you can script access to the API:
`curl --digest -u "user:password"  [url]`

Liniik, 2024-01-23:
One would expect they mention this in release notes.... I didn't update because it works and isn't broken but if I knew this, I'd update immediately

JeremyDWilliams, 2024-01-23:
> This should be closed, v5.1.2 uses digest authentication with a username and password.
> 
> I had to hit F5 to get Chrome to save the username/password.
> 
> Note that this means you can script access to the API: `curl --digest -u "user:password" [url]`

I don't feel that writing a script is exactly a solution. It helps. but leaves original issue unfixed. It's more like a work-around.

twshield, 2024-01-24:
> I don't feel that writing a script is exactly a solution. It helps. but leaves original issue unfixed. It's more like a work-around.

I said you *can* use it from a script, not that you have to.  The request above was to change to digest based username/password authentication that a browser will save and they have done that.

jadoglio, 2024-01-27:
The worst part is once you log into this useless site you have no ability to control the printer you just get some simple data outputs.  Pretty pathetic...

twshield, 2024-01-27:
> The worst part is once you log into this useless site you have no ability to control the printer you just get some simple data outputs. Pretty pathetic...

No, you can upload a gcode file and start it printing or print a file already on the USB drive.  Are you using the latest firmware?

rkrussell, 2024-02-02:
> No, you can upload a gcode file and start it printing or print a file already on the USB drive. Are you using the latest firmware?

You cannot pre-heat the tool or the bed. Also, the Prusa Slicer creates .bgcode files, but the upload tool recognizes only .gcode fies. Aggravating. And, yes, I have the latest.

jamiebegin, 2024-02-03:
> You cannot pre-heat the tool or the bed. Also, the Prusa Slicer creates .bgcode files, but the upload tool recognizes only .gcode fies. Aggravating. And, yes, I have the latest.

The web interface allows .bgcode uploads. You just have to change the file extension filter in the upload dialog to show all files, not just .gcode.

But I agree with the sentiment throughout this issue. This is a very poorly designed and built authentication system. All the ultra-high security of hard-coded credentials combined with the convenience of being unable to disable it. Truly the worst of both worlds.

kevinsaucier, 2024-05-02:
Just throwing an additional 2 cents on here.  I just finished building my Mk4.  Been using PrusaLink on a Pi Zero 2 on my MK3S+ for a while now and finally got used to not using Octoprint.  Booted up the MK4 and found this nonsense hardcoded name/password.  I don't give a crap how secure the PrusaLink interface is on my local network.  Why do I need to remember KeYZpcQjk6xuDJz for every time I log in?  That's just ridiculous!

And why is the Settings page not available on the MK4?  I have to use the tiny LCD to do everything?  Sigh.......Prusa makes some amazing printers but it's just amazing that they've committed what, 1, maybe 2 developers to what is a necessary part of 3D printing nowadays?  And still no phone apps to allow for management......    Sigh......

wafflecart, 2024-05-02:
> Just throwing an additional 2 cents on here. I just finished building my Mk4. Been using PrusaLink on a Pi Zero 2 on my MK3S+ for a while now and finally got used to not using Octoprint. Booted up the MK4 and found this nonsense hardcoded name/password. I don't give a crap how secure the PrusaLink interface is on my local network. Why do I need to remember KeYZpcQjk6xuDJz for every time I log in? That's just ridiculous!
> 
> And why is the Settings page not available on the MK4? I have to use the tiny LCD to do everything? Sigh.......Prusa makes some amazing printers but it's just pathetic that they've committed what, 1, maybe 2 developers to what is a necessary part of 3D printing nowadays? And still no phone apps to allow for management...... Sigh......

Are you on the latest firmware? There was a fix a while ago for username and password not working with password managers / auto fill. 

I find Prusa Connect (cloud) is great and has all the features you expect but unfortunately Prusa Link (local version) is so dumbed down I was actually surprised first time I booted up, I was like “this is it..”

wafflecart, 2024-05-02:
> The worst part is once you log into this useless site you have no ability to control the printer you just get some simple data outputs. Pretty pathetic...

Prusa Connect (runs in cloud) allows you to control XYZ of printer, heat up bed and nozzles, queue prints, start stop jobs, cancel objects on bed etc, it’s pretty decent, however as of now Prusa Link is a dumbed down version of it and is severely lacking 😞

kevinsaucier, 2024-05-02:
> Are you on the latest firmware? There was a fix a while ago for username and password not working with password managers / auto fill.
>
> I find Prusa Connect (cloud) is great and has all the features [...]

I am on the latest firmware and it does seem to save the password but it's still a crappy password.

To build on your comments, what really sucks is all of those controls are available in PrusaLink on my MK3S+ with a Pi Zero.  They just aren't available on the xBuddy version.  🤦‍♂️  And entering the URL directly on the xBuddy version just causes the page to save it hit an error.  Sigh......

[image]

mix579, 2024-05-02:
Unless you have some fundamental issues with cloud-based control, I see no reason to even look at PrusaLink. I run my small farm with PrusaConnect, and while there are certainly quite a few opportunities for improvement, it works very well.

kevinsaucier, 2024-05-02:
> Unless you have some fundamental issues with cloud-based control, I see no reason to even look at PrusaLink. I run my small farm with PrusaConnect, and while there are certainly quite a few opportunities for improvement, it works very well.

Well, you can see in the screenshot that my MK3S is not connecting to PrusaConnect for some reason (currently whining about not being able to get the serial number but that flip flops), so I do indeed have an issue with cloud based control.  🤷‍♂️  It really makes no sense to have to send all my data to Prusa only for them to send it right back to me.  I'd prefer to send it local and have it then get uploaded to Connect as a backup/extension.  Also, not that it currently affects me, but there is a space limitation on Connect, which means it will eventually cost more money to use Connect that Link.

jjshoe, 2024-07-08:
The bot auto closing this issue will add unnecessary noise for everyone who cares about getting this actually enahnced.

nomike, 2024-08-16:
Hi!

I want to add a few things here:

HTTP-Digest authentication, as already mentioned, is a very secure industry standard.

I would say the majority of Prusa Users are not expert system administrators.

People will not be able to figure out how to configure TLS properly. Most printers will not have a publicly available domain-name (more something like `prusamk4.lan` or something similar), so buying a TLS certificate from a publicly trusted CA won't be possible. Also you would have to pay for it annually. I don't want to expose my printer to the public internet, so let's encrypt isn't an option either.

Even less people will be able to setup their own local certificate authority.

So it's safe to say, that TLS is not an option. And just in case you really want that, nobody is preventing you from putting the printer in a separate VLAN, behind a firewall and setting up an nginx reverse proxy in front of it where you can have all the TLS mumbo jumbo your heart desires.

So in light of this, using HTTP digest is the most secure choice and far superior to sending passwords unencrypted through the network by using some custom HTTP-forms. And it's a  very mature and quite secure industry standard supported by all browsers.

As to the question about password managers:
I'm using KeePass2 as my password manager and I'm using the Kee plugin in both Firefox and Chromium/Chrome to connect to the database. Kee is perfectly capable of auto-filling the HTTP digest password prompt displayed by the browser.
So if you are using 1Password or some other service which can't handle these, I would rather recommend filing a bug report or feature request with those.
It's not Prusa's fault that some password managers seem to not get this sorted out.

But I'm a friend of taking care of my own valuable data, so instead of trusting some company to keep my passwords safe (and there have been reported data breaches with them over and over, which is no surprise as they are a very valuable target for hackers), I'm using the open source KeePass password manager. I store my encrypted database on a WebDAV folder, so it syncs between all my desktop devices (Linux, Windows, macos) and android phones flawlessly.

And regarding the auto generated password:

As I already mentioned, the majority of Prusa users will not have a background as sysadmins. So they will use weak passwords, will use passwords they re-use on many other systems etc..
Having auto-generated passwords prevents that.

And if you have a proper password manager, you only ever have to type it in once anyway. So IMHO it could be even more complex to further increase security.

And the argument about passwords being written down on sticky notes is not really a strong one. If you have physical access to the printer, you can do whatever you want. You can set the temperature to very high values, connect a USB drive to upload some rogue firmware or simply lookup the password in the settings. I'm not concerned about physical security.

I'm concerned though about network security and this is done quit well so far.

Regards
nomike

Prusa-Support, 2024-11-16:
Thanks for sharing your feedback.

This conversation was quite extensive and rich in valuable opinions and technical details that our developers appreciated.
Because the thread is rather long and detailed though, I may be missing something but, for the sake of conciseness, it seems safe to stick to the opening message and the summarized points.
>(1) PrusaLink must allow users to choose their own username [...].
>(2) PrusaLink [...] is not compatible with password managers.

1 . A user may have missed this issue and fortunately created a second more specific issue, which is proven to be more effective for the large majority of cases (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3559).
The other issue stayed rather silent but got our developers' attention as a consequence of the stale bot action which we do double check.
Our developers are working on this.

2 . This is probably relevant for Prusa Link Web repository where the topic was already addressed - https://github.com/prusa3d/Prusa-Link-Web/issues/465#issuecomment-1961500315.
I seem to understand that most common browsers should be able to save credentials but certain password managers will probably require tailor-made workarounds anyway.

.

> The bot auto closing this issue will add unnecessary noise for everyone who cares about getting this actually enhanced.

Or... Will make sure that even old issues, if still relevant, get more attention, and this is the case - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3161#issuecomment-2212781035. 🙂 
There are probably many other potentially obsolete and distracting issues out there so the bot is very necessary at the current time. About that, https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3559 is another example of how the action of the stale-bot should really have reason to be perceived negatively, whereas it helped draw attention to the issue even so quietly.

.

Sorry for the long message and I apologize in advance if I missed some important details. Please let me know.
Other than this, I think these requests could be more efficiently addressed as split in the other two issues mentioned above.
In this sense, the current issue could be closed, as it could feel otherwise somewhat overwhelming.

Michele Moramarco
Prusa Research

neuron-whisperer, 2024-11-19:
@Prusa-Support Thank you for the response. I need to address a few dangling threads:

> (1) PrusaLink must allow users to choose their own username [...].

I can live with the notion of not choosing my username. It's a peculiar choice, but a livable one.

The much more important part of suggestion (1) is the part that you edited out:

> and also allow users to specify their own passwords

There's still no way to do this. We're stuck with whatever alphanumeric soup the printer auto-generates for us. We can ask it to generate a different alphanumeric soup, but that doesn't help anything.

Look - I have two printers. Needing to store and use a unique alphanumeric password for one of them is a pain. Needing to store and use *different passwords for each printer* acutely amplifies the pain. I can't even imagine how printer farm managers cope with needing to manager (x) unique passwords for (x) printers.

The kicker is that if Prusa is going all-in on security as has been suggested, fixed-length, alphanumeric-only passwords are terrible! Allowing people to choose arbitrary-length passwords *and* with the expanded character set of special characters would greatly increase entropy. Please google "xkcd correct horse battery staple" and consider that post for more info.

Bottom line: **At the very least, please let users choose their own passwords.**

> I seem to understand that most common browsers should be able to save credentials but certain password managers will probably require tailor-made workarounds anyway.

I get this general feeling that Prusa is disinclined to consider password manager support as an important feature. And I'm saddened by it.

Look - **browsers are terrible password managers**:

The Chrome password vault is stuck in Chrome and not available outside of Chrome - it's a Google lock-in mechanism, another reason to discourage people from using mobile Safari, Firefox, Edge, etc.

Consider this: How does Prusa Connect work on iPhones? The Prusa App. Which isn't Chrome. So if I change my password on my workstation and Chrome saves the updated password, I lose access on my phone *until I manually type in the password*, which is an enormous pain. Password manager support would solve all of that.

For those reasons and others (e.g.: security, transparency, portability, etc.), "just use Chrome as your password manager" is not an acceptable solution here.

Password managers are an invaluable tool in a secure computing environment, and Prusa is standing in the way of that. Please, please encourage your engineering team to reconsider these choices.

jjshoe, 2024-11-20:
If being secure as possible is the goal, then let's just move to passkeys while we're at it.

p27182, 2024-12-24:
has anything, or is anything being done about this?  this is one of the most annoying aspects of printer upkeep.  having to go through a whole process of updating passwords in a password manager for the browser, then copying it into prusa slicer...  (apologies im frustrated dealing with multiple issues with my xl at the moment)

Prusa-Support, 2025-01-04:
> 1 . A user may have missed this issue and fortunately created a second more specific issue, which is proven to be more effective for the large majority of cases (#3559). The other issue stayed [...]

Again, please do check the above-mentioned issue.  Specifically -> https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3559#issuecomment-2149694877 🙏
Again, even though this may take some time, our developers are working on it. 🙂 

As far as I can see, my previous message still applies - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3161#issuecomment-2480582841 - and I would like to highlight the final part of it.
 
> Other than this, I think these requests could be more efficiently addressed as split in the other two issues mentioned above.
In this sense, the current issue could be closed, as it could feel otherwise somewhat overwhelming.

Michele Moramarco
Prusa Research

airtonix, 2025-10-06:
@Prusa-Support just remove the password feature all together.

If you have stakeholders that want to protect their printer web interfaces then make them setup an auth solution in front of their fleet. (nginx, traefik, caddy, etc)

For everyone else it's a poor half baked solution that doesn't hit the mark.

allanonmage, 2026-01-11:
Regarding the comment that Prusa owners are not sysadmins, that may be true at scale, but I'm a sysadmin (and developer and other stuff too).

I just got a Prusa XL and am exploring Prusa Link for the first time, trying to figure out how to log in. For something like this to be deployed in the 2020s is **_horrifying_**. Setting up a web page to use a self signed cert, TLS, and letting you create a username and password have been basic and trivial tasks for _over a decade_, even on embedded devices. Octoprint does this, though I realize a ARM architecture is probably way above and beyond the microcontroller they're using (that's another boneheaded decision IMHO), but Arduinos can run web pages (not sure about TLS), and can let you change a username and password. If they can put [Doom on a pregnancy test](https://www.creativebloq.com/entertainment/gaming/from-calculators-to-pregnancy-tests-people-cant-stop-running-doom-on-ridiculous-devices), Prusa can do the basics for a web page.

Given the fact that Prusa has an option for [high security environments](https://www.prusa3d.com/product/prusa-core-one-l-critical-infrastructure-edition/) and the ability to use encrypted USB sticks, that means there's someone over there with a security background designing stuff. The above might have been excusable as "that's not our wheelhouse", except for the fact that Prusa is pursuing high security clientele, and has acquired high security expertise.

This isn't just Prusa's fault: the security industry didn't exist, then as soon as it did, they didn't want to hire any actual talent, and now it's all corpo-liability dodging and insurance and stuff. I'm a cybersecurity hobbyist because I couldn't get a CISSP to get an entry level position (It's a paradox enforced by HR). _If you don't know, a CISSP is a management style certification that requires 5 years of provable cybersecurity experience, and sign off by another CISSP. This is why it's a Catch-22 paradox._

All that being said, TLS and usernames and passwords are super simple by the standards of the modern era.

If you think I'm being harsh, I am not. I come from a military and sports background, where we "are real" with our teammates and push them to improve. I have several Prusas, including an XL, which is a significant investment. I purchased it specifically to help with improving the community. So, this is me also helping with the community with more than my dollars.

reboot81, 2026-01-12:
Still awaiting an update, until then I'm not using PrusaLink.

jjshoe, 2026-10-16:
not stale