prusa3d/Prusa-Firmware-Buddy #3162: [BUG] "Print Complete" screen must be manually cleared before another print can start, and is sometimes shown repeatedly

neuron-whisperer, 2023-07-24:
**Printer type** - MK4 built from kit

**Printer firmware version** - 4.7.0

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint** - USB drive

**Describe the bug**

Upon completion of a print, the MK4 routinely displays a "Print Complete" screen. While this screen is being shown on the display, the printer is unable to start another print via the web interface - it results in some kind of "printer conflict" message in the web UI. The problem persists until the screen is manually cleared by selecting "Return" via the dial.

Potentially related problem: The "Print Complete" screen is sometimes displayed repeatedly. Selecting "return" causes the UI to switch to the model selection screen, but frequently, the UI spontaneously switches back to the "Print Complete" screen after a second or two. This behavior sometimes occurs multiple times after finishing a print.

**How to reproduce**

Print a model. After the model completes and while the printer is displaying the "Print Complete" screen, try to print another model.

**Expected behavior**

When the printer receives a request to print a second model while displaying the "Print Complete" screen for the first model, the printer should automatically escape the "Print Complete" screen and begin printing the second model.

Also: When the user manually clears the "Print Complete" screen, the printer should not spontaneously return to the same screen.

**G-code** - Not available

**Crash dump file** - Not available (no crash)

**Video** - Not available

neuron-whisperer, 2023-07-24:
> On a related ticket I saw a Prusa person remark that that's on purpose, to keep people from preventing sending a job while the previous model is still on the sheet. The logic is that you have to clear the model anyway and might as well clear the print finished message at the same time.
> 
> I get it but still feel there shouldn't be a need to babysit users to that extent. It clearly wasn't a big deal on the Mk3S. But I guess there might have been users on the Mk3 who tried to print over a model on the sheet and submitted this new feature as an "enhancement"--to the detriment of everyone else.

Interesting, and also peculiar. Is this a legitimate problem?

OctoPrint has no such requirement, and I've never seen any complaints about this as a missing feature.

neuron-whisperer, 2023-08-01:
I upgraded my printer to 4.7.1. With this firmware, when the web interface receives a request to start a print, it pops up a message to the effect of: "Are you ready to print? Is the bed clear?", and it starts printing after receiving confirmation.

I believe that this behavior occurs irrespective of what the LCD of the printer is showing, including the "Print Complete" message of a preceding print. Since this behavior supersedes the earlier behavior and does not require interaction with the printer, I presume that this issue has been resolved.

Still, I will mention that I find the message unnecessary and a little patronizing. I've owned 3D printers for years; I am aware that the bed must be clear before starting a new print. OctoPrint does not have such a warning, and I am not aware of any complaints that its absence is a deficiency. This warning should be optional and capable of being disabled.

mix579, 2023-08-01:
I support the idea of making this an option. A long time Astroprint added this warning to their UI but it quickly became optional when many of their users complained.

GRIFFCOMMca, 2023-08-21:
Seen this (v4.7.2) the screen remained on "printing" with "100%" i thought was a bug, at least "printing" should have changed to "complete" as i had no idea if the printer had finished.... The Mk3 didnt have this issue, not sure why its on the Mk4...

clivius, 2023-09-25:
This is super annoying, having to clear the print complete screen ON THE PRINTER in order to start another print from the web interface.  Web interface should know if a print is in process and or finished to either deny the upload or clear the print complete screen and start another print.

closed 2024-06-20 by github-actions[bot]