prusa3d/Prusa-Firmware-Buddy #3164: [BUG] PC-type C - printer resets.

Phardbean, 2023-07-25:
MK4-kit

Alpha4 Input Shaper Original

**Describe the bug**
  Well, turn your PC off and go to bed while your IS test print runs.  Wake up turn your PC on, which is still (printer running fine) hardwired through the TYPE-C and the printer resets.   Bye bye 9.5 hours of 12.

**How to reproduce**
  Please describe steps to reproduce the behavior.  ^look up^

**Expected behavior**
  I expect it to behave like my MK3s+ that is never unplugged ..., and never resets.

Phardbean, 2023-07-25:
name
Ulvheim_Merchant_House_Tier_01_C_0.4n_0.2mm_PLA_MK4IS_12h27m.gcode
state
UNKNOWN
print start
Yesterday at 10:55:03 PM
print end
Today at 8:00:22 AM
printing time
9 h 5 m 19 s
estimated printing time - normal mode
12h 27m 27s

shledge, 2023-07-27:
I wouldn't advise controlling any 3D printer directly with a PC via USB. Causes too many potential reliability issues.

Bpendragon, 2023-07-30:
@Phardbean That doesn't seem like a bug to me, that's how Prusa Printers react when plugged in, going way back. A new serial connection is detected, printer resets to boot and use the new connection. Same thing happens when you restart octoprint on a Mk3. Most USB ports shut down except potentially for power delivery when the computer goes to sleep, upon waking up the computer resumes operation and reconnects the USB ports that were DC'd during sleep. This reconnection is what causes the reset

@shledge Disagree, unless you specifically mean "PC" to talk about your laptop or desktop computer that you also use for everyday use. I've been running Octoprint on an RPi for YEARS across multiple printers without printing issues.

Phardbean, 2023-07-30:
> @Phardbean That doesn't seem like a bug to me, that's how Prusa Printers react when plugged in, going way back. A new serial connection is detected, printer resets to boot and use the new [...]

Okay, here we go. I am not controlling the printer via anything actively.  I have my mk3s+ plugged in for 6 months. No active control/monitoring/anything running I turn it on and off.  It has never reset the printer. I only ever connect pronterface to level the bed on the mk3. Was trying to set frequency to memory on the  4, but it seems only possible through custom gcode per print in printer settings atm. Now I am not certain because I haven’t ripped my pc out and I have not looked up the ports on the board.  But it could very well be I am plugged into passive ports?

Since the issue, I felt to the port next to the mk3 and out the mk4 next to it. Someone mentioned somewhere else it’s maybe a +v port.  Makes sense to me I guess. I will attempt to reproduce or not tomorrow.

shledge, 2023-07-31:
> @shledge Disagree, unless you specifically mean "PC" to talk about your laptop or desktop computer that you also use for everyday use. I've been running Octoprint on an RPi for YEARS across multiple printers without printing issues.

I specifically mean a desktop PC or laptop. A RPi with Octoprint is quite reliable.

Prusa-Support, 2023-09-29:
Thanks for reaching us.

This shouldn't really be called a bug but an intentional behavior, at least on MK3-series.
I actually find it odd that the USB connection doesn't reset your printer but that should theoretically be the correct behavior.

Please clarify or provide a video of the described behavior (on MK4 vs MK3) as rebooting should be expected.

Michele Moramarco
Prusa Research

Phardbean, 2023-09-29:
So this should be closed. Someone somewhere else point out I likely had the mk3S stuck in a no voltage slot, and the mk4 in a+ voltage slot.  This in fact was the case. Once I moved the the parallel port both printers are unaffected.

Phardbean, 2023-09-29:
Sometimes you plug it in, in the dark you know? It's little more fun.🤣🖖🤣

Prusa-Support, 2023-10-25:
Thanks for clarifying.
This issue will be closed now.

Michele Moramarco
Prusa Research

closed 2023-10-25 by Prusa-Support