prusa3d/Prusa-Firmware-Buddy #3165: [BUG] PrusaLink - Cannot create or delete directories

m-cas, 2023-07-25:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4]

**Printer firmware version** - [4.7.1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [na]

**USB drive or USB/Octoprint**
  USB flash drive

**Describe the bug**
I wanted to delete the two folders (buggy etc.) and create another one. No matter if the folder is full or not, the delete command will return an error. Creating a new folder errors out as well.

**How to reproduce**
  Try to delete or create a folder.

**Expected behavior**
  A folder will be deleted and created.

**G-code**
  na

**Crash dump file**
  na

**Video**
  na

Gordone47, 2023-08-18:
This problem also happens when attempting to upload a gcode from latest Prusa Slicer to PrusaLink.  "Error uploading to print host:  HTTP 500:  500 Infernal Server Error    Failed to Create Folder
[image]

drake127, 2023-09-19:
Creating directories don't work on 5.0.0 as well. I didn't test deletion.

drake127, 2024-05-28:
Hello, I'd like to get some confirmation from someone else but using 6.0.0 the issue seems to be fixed.

plug-it-in, 2024-06-20:
You cannot delete folder on 6.0.2

3d-gussner, 2025-09-22:
@nkohlmeier Please delete the files inside the folder. Also don't do these kind operations when printing.

nkohlmeier, 2025-09-22:
> [@nkohlmeier](https://github.com/nkohlmeier) Please delete the files inside the folder. Also don't do these kind operations when printing.

Thank you. I was able to successfully delete the directory after deleting all files in it.