prusa3d/Prusa-Firmware-Buddy #3166: [BUG] <Enter comprehensive title>

Phardbean, 2023-07-26:
**Printer type** MK4
**Printer firmware version** Alpha 4
**Original or Custom firmware** - Original
**USB drive and PL and Connect**
  Please let us know if you print via USB flash drive  couple times PLink comuple time PConnect

**Describe the bug**
Push print Z-raises 10-20mm and everything is frozen.

**How to reproduce**
  Keep printing it's happened once per day. past 3 days.  Usually after sitting on the print finished screen for a lone time,  why don't they time out?

**Expected behavior**
 Uh, not to freeze?

**G-code**
  ask for Gcode but don't accept the format?

**Crash dump file**
  It's printing.

**Video**
https://share.icloud.com/photos/03fDpWoLy_7vAOjzPvemiPdEQ

ericus256, 2023-08-12:
I am experiencing this issue as well. Happens about once every 3-5 prints. Exact same behavior- hit Print, z raises slightly, and then... nothing. Have to power-cycle. Happens with different USB drives, and extremely simple gcodes (hollow extrusion calibration cube & similar). Also running alpha 4. About to try 5.0.0 RC firmware.

ericus256, 2023-08-13:
Just happened again, running 5.0.0 RC. So looks like not fixed indirectly.

gudnimg, 2023-09-02:
I think I'm seeing this as well. 5.0.0 RC1

The image below shows the screen which is stuck. Everything is unresponsive. The knob click or scrolling does nothing. And the clock on the UI does not update. Is the system in a deadlock? The system seems to stay in this state forever so it seems the firmware is stuck in some loop.

I had finished a print and wanted to start another print (without power cycling the printer) but then this issue happened. I can confirm it did raise the Z-axis a few millimeters before getting stuck.

[image]

Linusten, 2023-09-04:
Exactly the same issue here aswell, running 5.0.0-RC

ericus256, 2023-10-17:
I just had this happen when sending a print via the slicer (PrusaLink) for the first time. Previously I was doing everything over USB transfer by hand each time, and after finally switching to WiFi, I had hoped that maybe this issue was gone. However, even over WiFi the same thing can happen apparently.

I mean, I guess technically PrusaLink is just pushing a file to the flash drive and starting the print remotely, but I guess it was just wishful thinking :)

closed 2024-06-20 by github-actions[bot]