prusa3d/Prusa-Firmware-Buddy #3174: [ENHANCEMENT] Add preset for 170degC PVA filament

ddavidebor, 2023-08-01:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - All buddy printers

**Is your suggested improvement related to an existing problem? Please describe.**
I'm using the ESUN PVA filament, which smokes at the toolchange temperature for PLA and cooks within the extruder.

**Describe the expected functionality**
When changing filament, in addition to all the current filament, a "PVA" filament should be provided (likely at the bottom of the list, to avoid confusion with PLA), with a  preset extruder temperature of somewhere between 170 and 200.

mdjurfeldt, 2023-10-24:
I agree. In addition to the problems listed above, it is strange to always get a filament mismatch warning when doing prints with soluble support.

ssill2, 2023-11-24:
Seems like a pretty significant oversight.  and I would suggest while they are at this, add bvoh too, though bvoh would work with the pva option for loading filament.

DoubleStrike, 2023-12-04:
Wanted to add another voice for this. Adding both PVA and BVOH would be handy since many of us got a multi-toolhead printer to do things like dissolvable supports.

ssill2, 2023-12-04:
it would also be nice if they would order the filament list by the ones you use most frequenly.  I hate having to scroll all the way down to the bottom to get to FLEX for instance.   There'd only ever be a handful of them clustered at the top if they did, at least for me.

ddavidebor, 2024-05-28:
NOT STALE!

ssill2, 2024-05-28:
agreed.  Having AT LEAST a PVA option, or better the ability to add custom options is needed.   I use both bvoh and aquasys 120 and it's always a kludge to load them. and then you have to skip past the warning that it doesn't think you have the correct type of filament loaded.

CZDanol, 2024-06-01:
Hey, I will try to push for adding support for custom material types. Unifying the related issues under the internal ticket BFW-5513.

closed 2024-09-11 by CZDanol