prusa3d/Prusa-Firmware-Buddy #3182: [BUG] Stuck on preparing bootstrap/looking for bbf

hacksawz12, 2023-08-07:
Printer type - MINI

Printer firmware version - 4.4.0

Original or Custom firmware - Attempting custom, but original with no modifications yet

Optional upgrades - Filament Runout Sensor

Describe the bug
  Trying to flash a non-modified version of firmware version 4.4.0, and after initial install it reboots to "looking for bbf". When I install the flash drive with the bbf it quickly says "preparing bootstrap", then alternates back to saying "looking for bbf". Then it just cycles back and forth between the two.

How to reproduce
  I downloaded the git repo, from branch release-4.5.0, built it, and tried to flash it via both dfu mode and via the flash drive, and got the same result both times.

Expected behavior
I need to be sure to have a working codebase before I can attempt any modifications to the code.

closed 2024-06-24 by github-actions[bot]

RobinSandhu780, 2024-07-16:
Hey I am having the same problom as you have you fixed it yet? If so can you tell me how you did???

Prusa-Support, 2024-10-15:
This issue is obsolete and probably closely related to old similar issues which are considered solved now.
Please check https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2862#issuecomment-1410251265 and https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2311#issuecomment-1917703235 and contact the official Customer Support if help is needed.

Michele Moramarco
Prusa Research