prusa3d/Prusa-Firmware-Buddy #3186: [BUG] Firmware too strict to fail hotend fan during preflight check

epiccitizen, 2023-08-09:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Mk4 Kit

**Printer firmware version** - 4.7.2
Was also present on firmware 4.7.1

**Original or Custom firmware** - Original

**Optional upgrades** - N/A

**USB drive or USB/Octoprint** - N/A

**Describe the bug**
Self-test intermittently fails hotend fan during preflight check

**How to reproduce**
 When performing preflight check with a new hotend fan, if the fan does not come up to the proper RPM speed in the expected amount of time, preflight fails. If you immediately re-run preflight, it will pass. The difference is that the fan was recently running. Monitoring the RPM of the fan and comparing against pass and no-pass situations shows the fan is just barely not spinning up quickly enough. We're talking within +-20 RPM or half a second sometimes. The fan works fine, and if the fan was recently ran it seems to spin up a little quicker.  Ignoring the error and using the printer, the fan seems to break-in and loosen up. After ten or twenty hours of use, it no longer fails preflight or hot-end fan self-test. This was reproduced on two separate hotend fans. 

**Expected behavior**
  Expected RPM value or failure delay window should be widened to allow for slightly more variation in fan performance. Either wait 2-3 more seconds for the fan to reach the expected speed, or allow a slightly lower RPM to be considered a "pass" 

**G-code**
N/A

**Crash dump file**
N/A one was not produced. 

**Images**
[  Please attach a video. It usually helps to solve the problem.](https://imgur.com/a/XMFn0qO)

tkircher, 2023-09-18:
Had this issue on my MK4 kit. The only part of the self-test that failed after running once. I upgraded from the provided 4.7.1 firmware to the 4.7.2 provided on the USB stick and thought that fixed it, because it passed, but now I'm wondering if that was just a coincidence. It was interesting because the fan definitely spun up correctly on visual inspection. There's a good amount of variability in the performance of these fans, even from the same lot.

Rex-Ignis, 2024-03-29:
I also have this issue with 2 of 5 Prusa mk4 printers running latest release but with the print cooling fans. However, mine do not pass unless I remove the fan and let it free float during the test. The printers have been operating for over 100 hours and replacing the fan does not fix the issue. I have also reprinted the fan mount which also does not fix the issue. They precisely fail because they do not reach the required rpm within the designated period. 100% a firmware issue. I hope this gets resolved because it is annoying to remove the fan when running self-tests.

danopernis, 2024-06-24:
This might have been fixed via commit https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/8828631d0794a0de17cf17d4f946cdc3c7de5feb since v5.1.0

Can you please confirm?

Rex-Ignis, 2024-06-24:
> This might have been fixed via commit https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/8828631d0794a0de17cf17d4f946cdc3c7de5feb since v5.1.0
> 
> 
> 
> Can you please confirm?

I can say that I am still having this issue as of most current firmware release.

closed 2024-09-02 by github-actions[bot]