prusa3d/Prusa-Firmware-Buddy #3191: [FEATURE REQUEST] Add Back Crash Detection

tbedore, 2023-08-12:
**Printer type** - [MK4]

**Is your feature request related to an existing problem? Please describe.**
  The MK3s/+ has crash detection on the X and Y axes I don’t see why its successor (MK4) doesn’t have the same feature. Unless I’m possibly missing a setting that I can enable but I cannot seem to find one.

**Describe the expected functionality**
Same as it would on the MK3 you could grip the X axis before the extruder hits into it and it would detect the crash, rehome the axes and then resume printing as if nothing happened. Again not sure as to why this feature didn’t simply transfer over to the successor

Ikario, 2023-08-12:
Yes, this is another big feature removed from the MK4 that should definitely be back.

Glaedr304, 2023-12-26:
I had to swap my filament during a print as the spool ran out, the end of the filament got jambed in the gearbox, and I had to manually take the filament out, during which I shifted the position of the x axis, I now have a ruined print because I cant rehome the x axis.

ernia93, 2023-12-27:
This feature would help a lot, especially when printing with many parts and difficult filaments. It is a shame that it is not implemented for the MK4. At least the user could be given a choice, as with the MK3. An increase in volume would also be totally acceptable for this.

johnkray, 2024-01-06:
I agree. If it's not possible to accurately do crash detection via the stepper drivers when also running input shaping, then they should have added closed-loop stepper motors. Not having crash detection has caused a lot of failed prints and wasted time/money. I hope they can either fix this with firmware and if not then with hardware on the MK4s and an upgrade kit for existing MK4.

Theheadtwo, 2024-02-18:
I've had so many issues with layer shifts that I'm tempted to revert back to MK3, this has gotten ridiculous.  I can't trust my MK4's to print reliably which is a big deal.  My XL has crash detection with input shaping, why not the MK4?  If it really is an issue with input shaping then make us disable input shaping to have crash detection on until you guys can figure it out please!

tbedore, 2024-02-18:
Exactly, I would very happily choose crash detection over input shaping, I rarely ever use input shaping. I’m surprised the XL has this I would assume it should be primarily the same system, only reason I could think it can’t be the same is if the XL has closed-loop steppers which I don’t think so? But I’m honestly not sure to that extent

tbedore, 2024-05-28:
Trying to keep this thing opened because I would like to see crash detection on the MK4

rcmaniac25, 2024-08-06:
Nothing like a 36h MMU print and the nozzle hits the transition tower and layer shifts with <5h remaining and keeps going... how is this not in the MK4? MK3 has it, XL has it (if phase stepping disabled... that's a different topic). But the MK4? Nah...

What happened?

Theheadtwo, 2024-08-07:
> Nothing like a 36h MMU print and the nozzle hits the transition tower and layer shifts with <5h remaining and keeps going... how is this not in the MK4? MK3 has it, XL has it (if phase stepping disabled... that's a different topic). But the MK4? Nah...
> 
> What happened?

The current XL firmware actually has it disabled now too... at least I can always revert it to get back crash detection unlike the MK4.  But yeah, this is a bit ridiculous.

rcmaniac25, 2024-08-07:
> The current XL firmware actually has it disabled now too... at least I can always revert it to get back crash detection unlike the MK4. But yeah, this is a bit ridiculous.

So that one is when phase stepping is enabled. If you turn off phase stepping, then crash detection works.

But on a bit of good news: I talked with support today about it, the individual had some misconceptions (we talked about the layer shift for the whole time... then at the end they said "we've determined you have a clog" which was not true. They also said I should run the MMU without input shaping, because crash detection on the MK4 is disabled when input shaping is used, and I had to inform them that Prusa's own config only offers the MMU3 on MK4 with input shaping, which surprised them). But once we got past that, they informed me that the engineers are still working on supporting input shaping with crash detection and they hope it will be a reality soon.

TheYang, 2024-09-17:
So, any official updates for this?  
nearly 5 months ago the Communication (for XL) was:  

> Currently, if Phase Stepping is enabled, crash detection is disabled. This is due to technical reasons. Developers are investigating whether these two features can work together.  

Is this investigation ongoing?  
Was it determined to be impossible?  
Was it determined to be too much work to prioritize?  
Where are we at?

tbedore, 2024-09-18:
From what I've seen online just now https://help.prusa3d.com/article/input-shaper-mk4-s-mk3-9-s-mk3-5-s-xl-mini_451816.

Crash detection is disabled due to high accelerations and print speed, crash detection feature is not compatible. 

Would love to actually hear somebody from Prusa reiterate this. But if I had to choose between having my printer go stupid fast or surviving a crash i'd personally choose crash detection.

TheYang, 2024-10-02:
XL can do crash detection with high acceleration though.  
but not with Phase Stepping.

How I See it:
Crash detection Availability

|| MK | Mini | XL  |
| --- | --- | --- | --- |
| No Input Shaping, No Phase Stepping | X | X | X  |
| Input Shaping, No Phase Stepping | | | X  |
| No Input Shaping, Phase Stepping | (X) | (X) |  |
| Input Shaping, Phase Stepping | | | |

rcmaniac25, 2025-01-18:
About 1/3 of the way into this print, the MK4S + MMU3 got caught on some purge that was sticking out of the purge tower. Instead of catching it with crash detection, it layer shifted and I canceled before it got back to the prints. Picture post-print cancel.

[image]

Avamander, 2025-07-06:
Just had a massive fail because a piece had warped slightly and the nozzle got stuck behind it and took most of the pieces on the bed with it. **Why** isn't this a feature on the MK4? 

At this point I'm considering making a "loud noise" detector with an Arduino or something to avoid these when I'm not in the immediate vicinity of the machine.